subsurface/subsurface #1924: Preferences > Units: Dive Time Units of Measure

seanwhe, 2019-01-02:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x ] New feature request
- [x ] Discussion request

### Issue long description:

The Units preferences do not address units of measure for time periods such as:
Bottom Time
Actual Bottom Time
Safety Stop Time
Decompression Stop Time
No Decompression Limit

It would be nice if the Time units group worked in the same way the Unit system group works. Enabling to personalise.

### Operating system:

### Subsurface version:

v4.8.4

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2019-01-03:
Which options would you like to have? All these decompression related times have resolution of one minute, so in my mind it does not make sense to show seconds there (which always would be 0). And in few cases, they reach time spans that would justify talking about hours.

For that matter, "bottom time" is an ill defined concept (at least for most dives), so I don't think we use it anywhere in Subsurface.

seanwhe, 2019-01-03:
Got you. That UI is confusing. The brain did not read the "Ascent/Decent denominator" label.

"Ascent/Decent denominator" should maybe be moved to the "Duration Units" option group. Removing the need for the "Time units" option group.

I agree the resolution is in minutes, my bad, the option provided for duration are perfect.

Thanks for your consideration.

mikeller, 2026-10-17:
Hi @seanwhe,

Following up on atdotde's explanation: bottom time, safety stop, decompression stop and NDL are all computed and displayed at one-minute resolution by design (seconds would always show zero at that resolution; hours aren't meaningful for these particular durations), and you'd agreed that made sense at the time.

Closing this per that agreement. Please reopen if you still want configurable units for a case where minute-resolution genuinely doesn't fit.

closed 2026-10-17 by mikeller