subsurface/subsurface #1925: Desktop Crash When Importing Dives to New Trip From Descent

UncleConker, 2019-01-03:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Desktop application crashes when importing dives into a new trip from the import dialog box but not when importing without a new trip.

### Operating system:
Microsoft Surface Pro 3
Operating system: Windows 10 (10.0)
CPU architecture: i386
OS CPU architecture: x86_64
Language: en-US

### Subsurface version:
Version: Subsurface 4.8.4.0
Does the same happen on another Subsurface version?: Unsure
Are you using official release, test build, or compiled yourself?: Official

### Steps to reproduce:
1) Hover over Import tab
2) Click Import from Dive Computer
3) Select "Garmin" and "Descent Mk 1"
4) Enter appropriate USB port
5) Tick "Download into new trip" option
6) Click "Download"
7) Select dives you wish to import
8) Hit "OK"

After a brief load and spinning cursor the application will crash.

### Current behavior:
Application crashes when downloading dives into a new trip but not when downloading without creating a new trip.

### Expected behavior:
Dives download successfully and create a new trip containing the new dives.

### Additional information:
**When I went to run the dump log the import was successful and created a new trip as expected the first time.**
[Log File 1](https://drive.google.com/file/d/1tpbxmFCW3lYjkrfIxGefShvz-Ik7x-CV/view?usp=sharing)

**The second time I ran a Log file the application crashed as well**
[Log File 2](https://drive.google.com/file/d/1tpbxmFCW3lYjkrfIxGefShvz-Ik7x-CV/view?usp=sharing)

Dump File: When I try to create a dump file I get a  "Dive Import Error"

### Mentions:

dirkhh, 2019-01-03:
A lot of that code has been massively rewritten, lately. Would you be willing to test a current development version to see if you can still see the problem there?
You usually can find our latest binary here: 
https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous 
just grab the subsurface-4.8.4-*.exe (as I write this it's https://github.com/Subsurface-divelog/subsurface/releases/download/continuous/subsurface-4.8.4-126-g0903bef9db86.exe but that will change once we merge the next pull request)

janmulder, 2019-01-03:
I tried to reproduce this crash, and failed. Obviously, I tried my OSTC3, tried from a clean logbook, and tried both master and 4.8.4 using the Appimage, on Linux that is. Simply, all just worked. And also the OP cannot reproduce at will (it seems).

UncleConker, 2019-01-03:
Jan,

Was this using the current official release or dev build? The only time it
was successful was the first time I exported a log, every time thereafter
the application crashed.

UncleConker, 2019-01-04:
Running the import with the latest dev build worked fine! I still received
a "Dive Import Error" when trying to create a dump file but I uploaded the
log file to Google Drive because it was too big to email.
<https://drive.google.com/file/d/1Fiyn1BLXfP_o0zkQp39R1-mrvTBsYezq/view?usp=sharing>

Hope this helps!

glance-, 2019-11-10:
Was this fixed with #1930?

UncleConker, 2019-11-12:
I have not had any recent dives that I can test on, I will have some next
month I can report back on.
Kind Regards,

Colin C. Baker
Preferred Pronouns: He, Him, His

bstoeger, 2020-05-03:
I suppose that this has been fixed with the commit referenced above, therefore closing. If this still breaks, please reopen.

closed 2020-05-03 by bstoeger