subsurface/subsurface #1926: Unable to connect & Download from Cloud Storage on RaspberryPi

keithgrimes, 2019-01-04:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Compilation of software completed on a RaspberryPi (as no existing distribution available). Software starts up ok. I can download software from my dive computer but am unable to connect to cloud storage  due errors raised (see below). 

### Operating system:
Debian Stretch
RaspberryPi
Have tried twice (including on a clean installation). Same error is reported. 
I also run 4.8.4 on iOS with no problems (but this is not a self compiled version). 

### Subsurface version:
Version Built: 4.8.4
Does not happen on pre-built versions of 4.8.4 (I run the packages on iOS & OSX without problem). 
Self Compile
Git Commit Ref used: 0903bef9db8659313c4bb533c295505c0c3851bf

### Steps to reproduce:
Execute the software. 
Configure the Cloud Storage account in Preferences
Select File>Open Cloud Storage

### Current behavior:
On first execution the error message ‘Creating Commit Failed’
On second execution the error message ‘unable to lookup revision <email>’ is displayed

### Expected behavior:
Dive Log already stored in the cloud is downloaded and displayed.

### Additional information:
Build Log available if needed. 

### Mentions:

keithgrimes, 2019-01-04:
Forgot to mention I have reviewed other entries with similar messages and tried deleting the directory ~/.subsurface/cloudstorage but this does not resolve the issue.

mturkia, 2019-01-05:
Have you installed all the dependencies listed in INSTALL file?

We do provide Debian packages from our Launchpad repository that should be compatible with Stretch, thus there should not be real reason to compile yourself just to get the software running.

Instructions to use the Launchpad .deb are available in https://subsurface-divelog.org/download/

keithgrimes, 2019-01-05:
As stated attached are build files, logs and an updated CMakeLists.txt file required to ensure the build completes (as instructed on an article on the User Forum). 

Please note two of the dependencies conflict (I have tried compilation with both). The other does not exist.

[config.txt](https://github.com/Subsurface-divelog/subsurface/files/2729374/config.txt)
[build.log](https://github.com/Subsurface-divelog/subsurface/files/2729375/build.log)
[CMakeLists.txt](https://github.com/Subsurface-divelog/subsurface/files/2729376/CMakeLists.txt)

janmulder, 2019-01-05:
Just a wild guess ... I see that you are using libgit2 0.24.0 ... the official Linux build is currently using 0.26.0. As the problems you see are related to git handling, it might be worth to give the newer libgit2 a try.

mturkia, 2019-01-05:
I forgot that Raspberry is ARM, so no packages from Launchpad. But I think Jan's assumption might be spot on. If that is the case, we should update the build scripts to compile libgit from sources also when 0.24 is detected.

keithgrimes, 2019-01-05:
Thanks. Will give it a go later today and let you know how I get on.

keithgrimes, 2019-01-06:
Jan/mturkia, 

Ok some success, but not completely. The latest version of libgit2 using apt-get on a RaspberryPi is 0.24. So I have taken the option of self compile the sources as part of the subsurface build. Using build 0.26.0 (as per your recommendation) removes the error messages previously noted - Great. However, now after the download has completed the dialog box stating ‘Successfully opened dive data’ remains on the screen with the progress bar at 100% and the application hangs. I cannot see any error messages being displayed. I have tried later version of libgit2 including 0.26.8 and 0.27.7. (latest) without success.

Do you have any suggestions on what this may be? or how I can find out more?

Logs/script below
[build_subsurface.txt](https://github.com/Subsurface-divelog/subsurface/files/2730398/build_subsurface.txt)
[build.log.txt](https://github.com/Subsurface-divelog/subsurface/files/2730399/build.log.txt)

Thanks for the support
Regards
Keith

mturkia, 2019-01-06:
when you start Subsurface, use -vv option. That will enable some debug outputs on the console. I just started a compile of Subsurface on a RaspberryPI, so in a couple of hours, I might know a bit more.

keithgrimes, 2019-01-06:
many thanks. Will do and let you know.

keithgrimes, 2019-01-06:
Just completed the run with -vv option and attached a log of stdout/stderr below. 

Not sure if this says anymore - but happy for the guidance. if you get your RPI version working, happy for you to use my Cloud details. 

Regards

Keith
[subsurface.log.txt](https://github.com/Subsurface-divelog/subsurface/files/2730468/subsurface.log.txt)

janmulder, 2019-01-06:
I see (at least) 2 issues. 1) There seems something weird with OpenGL version (what to try is in the log output).  2) the map widget plugin cannot be found (as its compiled with --skip-googlemaps), but we might expect that when things are compiled without maps, we do not get runtime errors from a missing map.

The verbose output further shows that the reading of the data from the cloud storage completed successfully (and read 32 dives, @keithgrimes: is that a correct number of dives?). So, @mturkia, we should fix the build for a newer libgit2 than 0.24.0.

Finally, Qt 5.7.1 is ... old.

keithgrimes, 2019-01-06:
Hi Jan, have tried setting the environment variable QT_QUICK_BACKEND as suggested with the error but that seems to make no difference. 

With regards the number of dives - 32 is correct.

Regards

Keith

janmulder, 2019-01-06:
Lets wait for @mturkia for more details ... I'm tempted to suggest that the old Ot plays a role in this next issue. While we do compile tests against older Qt's, and run automated tests on them, no UI related testing is done.

atdotde, 2019-01-06:
Hi everybody,

> On 6. Jan 2019, at 17:32, Jan Mulder <notifications@github.com> wrote:
> 
> Lets wait for @mturkia <https://github.com/mturkia> for more details ... I'm tempted to suggest that the old Ot plays a role in this next issue. While we do compile tests against older Qt's, and run automated tests on them, no UI related testing is done.

I just finished building Subsurface on a Raspberry Pi. This was hard enough since it kept dying probably from overheating in the process. I had do disable the GL version test (due to missing symbols) and build libgit2 manually (without any problems). But now I have it running, or better crawling but with operational maps and cloud interaction.

Best
Robert

atdotde, 2019-01-06:
> On 6. Jan 2019, at 18:22, Jan Mulder <notifications@github.com> wrote:
> 
> What OS @atdotde <https://github.com/atdotde>?

Plain vanilla raspian

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.6 (stretch)
Release:	9.6
Codename:	stretch

keithgrimes, 2019-01-06:
I am running the same version (no mods, built from NOOBS/Raspbian Full). 

I get the same output when running lsb_release -a. 

I have removed some items such as Wolfram, Sonic Pi, Libre Office to free up space. That may be the only difference. 

@Robert - Do you have a build script?

atdotde, 2019-01-06:
> Am 06.01.2019 um 19:08 schrieb keithgrimes <notifications@github.com>:
> 
> @robert - Do you have a build script?

Only subsurface/scripts/build.sh

And, as i Said i had Tor estate the Build a number of Times including running make clean in between and commenting out some GL calls. 

Robert

keithgrimes, 2019-01-09:
All, many thanks for the help so far. After spending some time reviewing the build I have now managed to get the maps working and subsurface compiled with libgit2 version 0.26.0. This has resulted in the download of my dives now working successfully with the performance on my Pi 3B+ is not too bad.

FYI - The reason the MapWidget did not find the googlemaps plug in was due to the path being incorrect (or too long) in the install statement. I separately installed the file into /usr/lib/arm-linux-gnueabihf/..... Please see the build log for more details. 

However, I am left with 1 (hopefully final) challenge. When I change something (change the visibility or add a further comment) the upload to cloud storage fails with an error in the UI of 'Creating Commit Failed'.  I have looked in the log files but struggle to get an idea as to why?  I have tried a later version of libgit2 (0.27.7) but that failed (with very different messages). 

Logs (build and execution) are attached. Any hints and tips would be gratefully received? I assume they is a Git Commit but do not understand why this could fail. The only line in the log I think should not be there is 'removed reference to non-existant dive site with uuid 04e60857' but don't know what would cause this either. 

Once again many thanks for the support given to date. 

[output.log](https://github.com/Subsurface-divelog/subsurface/files/2742611/output.log)
[build.log](https://github.com/Subsurface-divelog/subsurface/files/2742615/build.log)

janmulder, 2019-01-09:
@atdotde: can you verify if you can save to cloud successfully on your build?

atdotde, 2019-01-10:
Keith,

> On 10. Jan 2019, at 06:26, Robert Helling <helling@atdotde.de> wrote:
> 
> trying once more, I can also save to the cloud properly.

could you try to save to a local git repository? You can do this following these steps:

On the command line, create a directory and in that directory run git init.

In subsurface, start a new log, create a some dive (for example using the planner or add dive) and then do “Save as”. Go to your new directory and choose a file name which is a string in square brackets like [foo].

Go back to the command line. foo is supposed to be a branch name which is not checked out by default, so do it explicitly: git checkout foo. Check if your directory contains your dive data.

BEst
Robert

keithgrimes, 2019-01-10:
Hi Robert, 

Following your instructions I get as far as saving to the new directory. I used the filename '[foo]' as you suggested and I get the error 'Creating Commit Failed' in the UI - Same as saving to Cloud. I tried both imported data and a single dive created by just pressing 'add dive'. Both provide the same message. 

Regards

Keith

keithgrimes, 2019-01-10:
If it helps the console log states: 
pi@pimobile:~/Documents/subsurface/src/install-root/bin $ ./subsurface -vv
libEGL warning: DRI2: failed to authenticate
using qt5ct plugin
Subsurface v4.8.4-141-g123f3ef7ec8f,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (e4c96e93caf68cf59e601da798e5dbaaf5bad60e)
built with Qt Version 5.7.1, runtime from Qt Version 5.7.1
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): WARNING: Detected OpenGL ES!\nAttempting to run with the available profile!\nIf this fails try manually setting the environment variable\n'QT_QUICK_BACKEND' with the value of 'software'\nbefore running Subsurface!\n"
can't find Qt localization for locale "en-GB" searching in "/usr/share/qt5/translations"
Plugins Directory:  QDir( "/home/pi/Documents/subsurface/src/install-root/bin" , nameFilters = { "*" },  QDir::SortFlags( Name | IgnoreCase ) , QDir::Filters( Dirs|Files|Drives|AllEntries ) )
D-Bus global menu: no
JIT is disabled for QML. Property bindings and animations will be very slow. Visit https://wiki.qt.io/V4 to learn about possible solutions for your platform.
loading dive data from ("/home/pi/.subsurface/username.xml")

File locations:

cloud URL set as "https://cloud.subsurface-divelog.org//git/keith@kcgrimes.co.uk[keith@kcgrimes.co.uk]"
Local git storage: /home/pi/.subsurface/cloudstorage/2dc600a00fd14e49
Cloud URL: https://cloud.subsurface-divelog.org//git/keith@kcgrimes.co.uk[keith@kcgrimes.co.uk]
Image filename table: /home/pi/.subsurface/hashes
Local picture directory: /home/pi/.subsurface/picturedata/

git storage: do git save
git storage: Preparing to save data
git storage: Start saving data
git storage: Start saving dives
git storage: Done creating local cache
git storage, write git tree

atdotde, 2019-01-10:
Keith,

> On 10. Jan 2019, at 20:59, keithgrimes <notifications@github.com> wrote:
> 
> Following your instructions I get as far as saving to the new directory. I used the filename '[foo]' as you suggested and I get the error 'Creating Commit Failed' in the UI - Same as saving to Cloud. I tried both imported data and a single dive created by just pressing 'add dive'. Both provide the same message.

now, I am starting to think this might be something with the content of your log file. Would you mind exporting your log as xml (possibly with annoymization turned on) and sharing that with us?

Best
Robert

keithgrimes, 2019-01-10:
Robert, 

Console log attached. For clarity created using the command line 'stdbuf -oL -eL ./subsurface -vv &> output.log' ensuring the buffer is only line cached. 
Instructions followed included 
1. Start the app
2.Select New Log from the File Menu
3.Select Add Dive from the Log Menu
4 Select Apply Changes (no modifications made)
5. Select SaveAs from file menu 
6. Enter filename of '[foo]' and select 'Save'

Log as attached. If there is another one, let me know. 
[output.txt](https://github.com/Subsurface-divelog/subsurface/files/2746993/output.txt)

atdotde, 2019-01-11:
From what's in the log, it looks like saving succeeded, there is no error recorded there. Can you check on the command line if there is something in the directory (remember you need to checkout branch foo first)?

I am currently at work and don't have access to my RPi, but here on my desktop, I can open your dives file and save it to a git repository, so it seems, there is nothing in your log, that interferes with saving to git. Weird.

keithgrimes, 2019-01-11:
Alas I am also at work but do have my Pi with me and will be able to access during lunch so will have a look. I am also thinking about adding some further verbose logging into save_git.c where the error appears to being generated from in order to understand which call is causing the issue.  Will let you know soonest.

atdotde, 2019-01-11:
This sounds like the creation of the repository got interrupted somewhere in the middle of the process. @dirkhh do you have any further idea how to debug this?

keithgrimes, 2019-01-11:
After some further analysis, and increased logging I have found the issue with cloud storage working for read and save 👍 . 
For reference the issue was that the account being used (pi) did not have any details set using the unix chfn function (in particular the full account name) and hence limited information in the /etc/passwd file. This only returned ,,,, which was invalid (hence the error for no user name configuration). Once set it worked fine. Now I just need to rebuild without all the logging information I have added.
Thanks for all the help. The support has been great for someone who has not developed in C for a number of years (and even then not on Linux).
Thanks very much.

atdotde, 2019-01-12:
Excellent debugging! I submitted a patch that should prevent ending up with an empty user name. Could you please check that this does the job?

keithgrimes, 2019-01-12:
Will do later today.

Regards

Keith

PR #1934 (2019-01-16): Don't allow empty username for git
When no real name is set in /etc/passwd the username ends
up being the empty string. Git does not like that.

Before we only testing for the username being a NULL pointer.

Fixes #1926 

Reported-by: Keith Grimes

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Release note:
closed 2019-01-16 by bstoeger