subsurface/subsurface #1952: Implement hide-filter functionality

bstoeger, 2019-01-26:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [x] Discussion request

### Issue long description:

The old filter widget could be hidden - it would then only show a single line "x of y dives shown".

The new filter widget can only be closed, whereby filtering is reset. It would be nice to have a similar hiding mode, because currently the filter hides the map and it might be useful to see e.g. all dive-sites with wall dives, etc.

### Operating system:

Any desktop OS.
### Subsurface version:

Master.
### Steps to reproduce:

CTRL^F
### Current behavior:

No way to hiding the filter, while keeping filtering active and showing the number of filtered dives.
### Expected behavior:

Some way to do just that.
### Additional information:

### Mentions:

@dirkhh @janmulder @tcanabrava

dirkhh, 2019-02-18:
I'm looking at implementing this - and there's a truly trivial way to do it, but I'm not 100% sure that's what we want.

Right now, when the filter widget is hidden, it clears the filter.
I can turn that off (one liner) and the filter stays active when you toggle between it and the map using Ctrl/Command-F (or when you close it with the close button).

As a result, the only way to get back to an unfiltered dive list is the 'reset' button on the filter pane.

Is that an ok behavior? It feels more natural than what we have today and I like it better than a separate 'hide' button. But I'm not sure how others would feel... @bstoeger @janmulder @atdotde

atdotde, 2019-02-18:
Sounds right to me. We could even have a small filter icon on the map when the filter is active to reset or reopen the filter.

janmulder, 2019-02-18:
You know me: I like hidden functionality :-) The solution without a "reset" button is perfectly fine for me, but I'm sure some people would like a button better to give a visual queue of the functionality.

bstoeger, 2019-02-18:
Yes, I think there should be *some* indication of an active filter, even if it is subtle. Perhaps only show the "x out of y dives line"?

dirkhh, 2019-02-18:
I'll need to move that indicator line out of the filter widget and place it somewhere else.
Something to noodle on, I guess :-)

commit f2dac58c (2019-02-18): Desktop: don't clear the filter when the widget is hidden

Only clear the filter when the user explicitly resets it. This way the user can
toggle between the map and the filter widget without losing the filter state.

Fixes #1952
commit e7f3cf13 (2019-02-18): Desktop: don't clear the filter when the widget is hidden

Only clear the filter when the user explicitly resets it. This way the user can
toggle between the map and the filter widget without losing the filter state.

Fixes #1952
commit 208b3bb9 (2019-02-19): Desktop: don't clear the filter when the widget is hidden

Only clear the filter when the user explicitly resets it. This way the user can
toggle between the map and the filter widget without losing the filter state.

Fixes #1952
closed 2019-02-19 by dirkhh