subsurface/subsurface #1959: Subsurface 4.8.5 bluetooth problem / Quad Air Mares

EBELLET, 2019-02-01:
### Describe the issue:

- [?] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Hello;
I am having a problem with subsurface synchronization with my Mares QUAD Air. It worked well with version 4.8.4. I'm tranmting the log file.

Can you help me? Does the problem come from me?

Thank you for your comeback.

### Operating system:

### Subsurface version:

<official release-->

Subsurface.log :
[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/2821047/subsurface.log)
Subsurface: v4.8.5, built with libdivecomputer v0.7.0-devel-Subsurface-NG (072bef16666e0ec397b8236e92722b2f1347b59b)
INFO: Open: transport=32
INFO: Configure: baudrate=115200, databits=8, parity=2, stopbits=0, flowcontrol=0
INFO: Timeout: value=1000
INFO: DTR: value=0
INFO: RTS: value=0
INFO: Purge: direction=3
INFO: Write: size=2, data=C267
INFO: Read: size=0, data=
ERROR: Failed to receive the answer. [in /win/subsurface/libdivecomputer/src/mares_iconhd.c:176 (mares_iconhd_transfer)]

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-02-01:
Clearly Subsurface is trying to connect to a serial interface here. Are you selecting bluetooth download? Is the dive computer paired / visible from the Windows BT tools?
@neolit123 am I missing any relevant questions?

neolit123, 2019-02-01:
these seem like the right questions.
i do no know if this DC is BTLE or regular Bluetooth and googling for that seems oddly hard. found this link:
https://www.houseofscuba.com/product/com222.html
so i'm going to assume BTLE.

@EBELLET next to all of the above questions could you please share a screenshot of your Import -> Import from DC dialog?
also make sure to read the user manual with the section about Bluetooth / Windows.

EBELLET, 2019-02-03:
Hello;

Thank you for your feedback. I redid the tests by cutting the blutooth of my phone .... ;-) .. I always have an error but not the same. I can sync Subsurface (Android version 2.1.6 (4.8.4.66)) with my phone but not with my PC. There is an element that can be impornant : my computer is brand new and has a single dive whose date is March 3, 2019 (error update date on my part). On the phone Subsurface informed me that there is no new dive to sync, on my PC I have an error message see the attached screen printing. my computer to repeat the test, I have the same error).

[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/2825525/subsurface.log)

[2019 02 03 - mares quad air blutooth pb](https://user-images.githubusercontent.com/47242450/52177097-93e36900-27bc-11e9-94b7-9a5fca09272e.PNG)

Microsoft Windows [version 10.0.17134.523]

Bests regards..

neolit123, 2019-02-03:
downloading on Windows is a bit tricky and there isn't much we can do about it.
we already mention in our manual that on Windows you might have to unpair/pair before each download.

some things you can try:

- go to start menu -> Bluetooth and other devices.
https://i.stack.imgur.com/yGJ6Y.png
- make sure your device is present in the list with status "Paired" and Bluetooth is "ON".
- keep this window open
- open subsurface and navigate to the screen where you want to perform the download but don't start it yet.
- go back to the "Bluetooth and other devices" window and remove your device from the list
- discover the device again and pair it
- put the device in "BT / Download mode" (the DC should have a menu for that)
- go back to subsurface and start the download
- if you look at  "Bluetooth and other devices" the device status should change from Paired to Connected.

your Windows version seems new enough.
in the future always keep it up to date, because Microsoft push BT fixes.

if a problem occurs again, it could be:
- a Windows Bluetooth driver problem. do you have the latest drivers installed from your vendor?
this fixed downloads for me.
- it could be a libdivecomputer problem (the library we use for downloads)

EBELLET, 2019-02-04:
Hello,

Thanks for these informations. I installed the new dell bluetooth drivers. I redid the procedure according to your message.

The connection is vbien, but I have the same error message and the log is different (see below).

Good reception.
[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/2828460/subsurface.log)

EBELLET, 2019-02-04:
Good evening;

I am sending you the result of a new test. Without the "LE", same error ..... I joined the log file.

Good reception and good evening ..
[image]
[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/2829150/subsurface.log)

neolit123, 2019-02-04:
oddly it seems that this isn't starting bluetooth communication at all. using serial instead.

i do not know what might be causing that. could be a problem in our library for talking to DCs - libdivecomputer.

EBELLET, 2019-02-05:
ok, I am ready to do new tests to help you.
Thank you for your time.
Come back to me if I can be useful.

Very sincerely

Eric

dm9876, 2019-03-12:
Hi all, I comment here as maybe my issue has the same root cause. I'm on Win10 and using Shearwater Petrel 1 (uses Bluetooth classic as its only way of download). I find a error with attempting download since 4.8.5 and find that reverting to 4.8.4 eliminates the problem.

**Steps**:
1. Ctrl+D
2. select shearwater/petrel and choose "choose bluetooth download mode" checkbox.
[image]

3. select the ... to the right of that and scan (while DC is in download mode)
the petrel is detected like normal. 
4. click 'save'
[image]

5. click download

6. Error dialog 
[image]

If I enable the subsurface.log it has limited info, just this one line:

Subsurface: v4.8.5, built with libdivecomputer v0.7.0-devel-Subsurface-NG (072bef16666e0ec397b8236e92722b2f1347b59b)

neolit123, 2019-03-12:
hi, i cannot confirm a problem on 4.8.5 with the DC devices that i have in classic Bluetooth mode.
could be a case where we broke something with respect to the Petrel between 4.8.4 and 4.8.5.

neolit123, 2019-03-12:
@dm9876 please try removing the DC from Windows 10's "Bluetooth devices", then re-add it and try again.

dm9876, 2019-03-26:
Hi, to close out my hijacking of this bug, all seems ok now with the Shearwater Petrel. I did try removing the device from Windows, and try again. It didn't work at first, but then a later date (a new dive and several restarts later) it all seemed to work smoothly with my Petrel via bluetooth classic on Win10 and Subsurface 4.8.5 not sure what the issue was.