subsurface/subsurface #1963: No images in profile export

atdotde, 2019-02-02:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2019-02-02:
That was my suspicion as well. I tried waiting for three seconds after creating the widget. That should have been more than enough time for the thumbnails to load. That did not help.

bstoeger, 2019-02-02:
I'm stupid - I had the picture thumbnails turned off. 😛

bstoeger, 2019-02-02:
> I tried waiting for three seconds after creating the widget. 

How? I think if you simply call `sleep()`, Qt's main loop will not run and the UI thread can't update the thumbnails. You would have to pass control to the main loop for a certain amount of time. IIRC, there's a special process signals call for QThread.

dirkhh, 2019-02-02:
We have code to do this the right way in quite a few places. E.g. `CheckCloudConnection::checkServer()`

bstoeger, 2019-02-02:
I chose to implement a "synchronous" mode, which will not do video thumbnail extraction or download files. This should not be a problem, as the thumbnails should have been created on addition of the files. 

So far so good. But now if you export multiple dives, they will all show the same thumbnails on different profiles. Will look into that tomorrow.

atdotde, 2019-02-02:
Indeed, sleeping with the event loop running does the job. Now, we need to figure out how to send a signal once the last thumbnail has loaded (or we give up).

commit 8a8063c1 (2019-02-07): Profile: add "synchronous" mode for picture plotting

The thumbnails were fetched in the background to achieve a
snappier UI. The problem with that is that on LaTeX etc.
export only placeholder thumbnails were shown.

Therefore, implement a synchronous mode. This only tries
to fetch cached thumbnails or calculate thumbnails for
images. Videos and remote files are not supported.

Fixes #1963
commit a16cca1f (2019-04-08): Profile: add "synchronous" mode for picture plotting

The thumbnails were fetched in the background to achieve a
snappier UI. The problem with that is that on LaTeX etc.
export only placeholder thumbnails were shown.

Therefore, implement a synchronous mode. This only tries
to fetch cached thumbnails or calculate thumbnails for
images. Videos and remote files are not supported.

Fixes #1963
closed 2019-02-07 by atdotde