subsurface/subsurface #1967: Subsurface for Fedora 29 with old dependencies

neufeind, 2019-02-07:
The RPMs for Fedora 29 contain a dep to qt5-qtlocation with a API private_5.11.1. But the current API-level is 5.11.3 with updated packages. Is there maybe a way to get rid of exact sub-patchlevel-versioning or (for now) trigger a release for the updated private API?
Currently trying to install subsurface would try to downgrade quite a number of qt5, kf5 and plasma-packages.

neufeind, 2019-02-07:
quite similar to #924

dirkhh, 2019-02-07:
The problem is that the distros have stopped packaging the private headers. I'm sure this was well intended, but we NEED those private headers. And Qt unfortunately does make ABI incompatible changes in those headers even in minor releases. So we have to package them this specifically.

I don't know how to solve this, TBH. Best solution would be for the distros to go back to simply packaging those headers so we could depend on the distro RPM. Other than that? No idea.

neufeind, 2019-05-06:
As an example I just looked that up in a package for Fedora 30:
https://fedora.pkgs.org/30/fedora-x86_64/qt5-qtlocation-devel-5.12.1-1.fc30.x86_64.rpm.html
Do I see right that might be the private headerfiles? Otherwise, I can happily true to reach out to the Fedora-folks, if you could provide me some hint what we'd need or where we're looking for a good solution from them.

dirkhh, 2019-05-07:
Maybe they added them back? I'll install Fedora 30 in a VM and try it out. If I don't update this issue in the next two days, please ping me again (because that likely means that somehow this has fallen off my execution stack... oops)

dirkhh, 2019-05-07:
Fun! Fedora is packaging the private headers again. I verified that `Subsurface` builds correctly on Fedora 30 without manually downloading the private headers.

Now I'll need to remember how to update the build specs so that our build picks those up correctly... I **think** this should work automagically, but... yeah, when do these things ever work the way you expect them to. But yeah, this is good news.

dirkhh, 2019-05-07:
The openSUSE Build Service hasn't added Fedora 30, yet. Until then, the easiest way to do this might be to build it from source yourself :-(
```
sudo dnf install git gcc-c++ make autoconf automake libtool cmake \
        libzip-devel libxml2-devel libxslt-devel libsqlite3x-devel \
        libudev-devel libusbx-devel libcurl-devel libssh2-devel\
        qt5-qtbase-devel qt5-qtdeclarative-devel qt5-qtscript-devel \
        qt5-qtwebkit-devel qt5-qtsvg-devel qt5-qttools-devel \
        qt5-qtconnectivity-devel qt5-qtlocation-devel libgit2-devel redhat-rpm-config
git clone git://github.com/Subsurface-divelog/subsurface
bash -x subsurface/scripts/build.sh -build-with-webkit
```
You can then start Subsurface as `./subsurface/build/subsurface`

neufeind, 2019-05-07:
I've tried to check with OBS, and to be use even asked on their chan on freenode: F30 is a supported build-target it seems. The list of targets is also here (but as a manual wiki-page):
https://en.opensuse.org/openSUSE:Build_Service_supported_build_targets

dirkhh, 2019-05-08:
...and I found how to enable Fedora 30 (I need to re-learn this every time a new release comes out, it seems). 
F30 packages are now available at the openSUSE Build Service

neufeind, 2019-05-08:
Works for Fedora 30, thank you.
Those private headers seem to be present in the devel-packages even down to Fedora 27 (lowest which I checked). Does that help you with automatic building maybe?
https://pkgs.org/download/qt5-qtlocation-devel

mikeller, 2026-10-17:
Hi @neufeind,

Thanks for working through this together. As you confirmed: "Works for Fedora 30, thank you." Fedora re-added the private Qt headers we need in Fedora 30, and we enabled the Fedora 30 build target on the openSUSE Build Service to match.

Closing this as resolved and confirmed. Please reopen if a current Fedora release drops those headers again.

closed 2026-10-17 by mikeller