subsurface/subsurface #1969: Importing with transmit error - Aeris Elite T3

martinatime, 2019-02-08:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I have a bunch of dive logs that I have exported from the Aeris Dive software into the ZXU format. Some of these files have data points where the air integrated transmitter failed to connect with the dive computer and therefore there is a point in the log such as
|5.00|511|||T|T|70.0|73||2912|
This skews the entire dive log and the overall dive statistics/profile. I'll attach an entire dive log file but here are some of the data points for context.
|3.50|19|||||0.0|71||2944|
|4.00|16|||||0.0|70||2928|
|4.50|18|||||0.0|70||2912|
|5.00|511|||T|T|70.0|73||2912|
|5.50|511|||T|T|70.0|76||2912|
|6.00|18|||||0.0|76||2888|
|6.50|19|||||0.0|76||2862|

I've switched dive computers (one that I can import directly into subsurface) but I would really love to be able to get all of my historical dive logs into subsurface. Thanks.

### Operating system:

Windows 10 Pro (1803) - English although I have tried this on Mac OS with the same results

Aeris Elite T3 - dive computer
MSI Laptop
MacBook Air

Yes

### Subsurface version:

4.8.5

I've tried this on previous versions

official release

### Steps to reproduce:

1) Click Import
2) Click import log files
3) Navigate and select the ZXU file
4) Click OK in the import dive log file dialog box
5) Dive profile now shows depth spikes to 511 feet or 156 meters

### Current behavior:

Dive profile now shows depth spikes to 511 feet or 156 meters after importing a dive log with a transmission error.

### Expected behavior:

Either transmission errors are ignored during import or they are averaged out or they are flagged and don't skew the dive profile and statistics

### Additional information:

[Aaron Martina_31.ELITE T3..ZXU.txt](https://github.com/Subsurface-divelog/subsurface/files/2845912/Aaron.Martina_31.ELITE.T3.ZXU.txt)

### Mentions:

martinatime, 2019-02-12:
I just noticed another dive that I loaded with a signal disconnect set the depth at 0. I didn't catch it before but I think it should be handled in the same fashion as the 511 feet.

mturkia, 2019-02-12:
I wonder what is going on with this log. When you have the depth spikes, water temperature keeps changing, and that seems to make sense. However, depth is clearly bogus and cylinder pressure is frozen. You also get ascent violation and decompression violation at the same time along with ceiling of 70 feet.

It really sounds weird if the DC is recording bogus depth in case connection is lost to pressure transmitter. What will happen when the transmitter runs out of power?

Anyway, we generally trust the data we get from dive computers and do not mess with it. And IMO it is not really a bug in Subsurface if we show correctly the data we are given. But I do agree that we could ignore the clearly bogus values. The question just is, what is considered bogus, what is correct threshold. Your log has 510 and 511 feet as bogus, but people do dive at such depths, and some dive computers have have a long sample interval and one might drop tens of feet during this time, ending up in 510 feet.

A quick fix you could do yourself would be to grep out the offending lines from the logs and import after that. On Linux (and I believe Mac) you could do the following:

for file in *.ZXU.txt; do grep -vw -e 511 -e 510 $file > ${file/ZXU.txt/mod.zxu}; done

This will loop through all the .ZXU.txt files in the current directory, remove lines that contain 511 or 510 and save the result to .mod.zxu. I have not tested this on Mac, but as the Mac's terminal uses bash as shell, this should work the same way it does on Linux. (Same can be done in Windows with powershell or findstr con command prompt, but someone else needs to figure that syntax out.)

martinatime, 2019-02-12:
@mturkia thanks for the response. You make valid points. In the Aeris software it shows those data points as link failures and seems to ignore them from a dive profile standpoint but they are counted for the overall dive time.

As I'm not familiar with the details of the ZXU file format, if I just remove lines/datapoints then won't that skew the profile as well? I know I can take the time to manipulate my own log files but I was hoping that I could contribute my issue back to Subsurface in the case that others have come across this.

Side note: I have a different computer now so I shouldn't have these types of issues going forward.

mturkia, 2019-02-13:
IIRC the first field in the DL7 (ZXU) format is timestamp and thus removing some lines from the log does not affect the dive duration. In Subsurface that would just mean that there is one or two points missing from the profile, and there the time between the previous and next valid depth points is that much longer. I will still check DL7 specification to make sure, but when I tested the removal of the offending lines from your sample log, the dive duration remained the same between the original and modified dives.

martinatime, 2019-02-13:
Do you happen to have a link to the DL7 specification?

mikaellindstrom, 2022-08-08:
I'm having a similar problem when importing directly from my Aeris Elite T3 - some data samples have invalid depths. didn't realize they may be due to transmitter link failures.
I have 2 second sample collection interval 
I wrote a small python script to clean up the .ssf file. I have 2 second sample collection interval and used a 1m depth different from the previous sample value as indication that it was incorrect and just copied the previous depth value into this new sample value.
I realized later that the meta data for that dive have to be changed too as the max and mean depth are recorded there too...

Example:
https://en.divelogs.de/dive/3397567