subsurface/subsurface #1971: Appimage depends on libidn.so.11, doesn't work under Void Linux

Veyrdite, 2019-02-10:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Subsurface appimage depends on libidn.so.11 to exist on user's system.  This doesn't necessarily exist -- my distro only has libidn.so.12.

Not familiar with the project, but I'd assume this lib should be shipped inside the appimage itself or used in a more flexible manner.

### Operating system:

Distro: x86_64 (glibc) Void Linux
Uname: Linux Shyarenris 4.19.18_1 #1 SMP PREEMPT Sat Jan 26 12:24:48 UTC 2019 x86_64 GNU/Linux

```
$ ls /usr/lib/libidn*
/usr/lib/libidn.so.12@
/usr/lib/libidn.so.12.6.0*
/usr/lib/libidn2.a
/usr/lib/libidn2.so@
/usr/lib/libidn2.so.4@
/usr/lib/libidn2.so.4.0.0*
```

### Subsurface version:

Release: Subsurface-4.6.4-x86_64.AppImage*

### Steps to reproduce:

* Use x86_64 glibc Void Linux
* Download x86_64 appimage from the front page of https://appimage.org/
* Attempt to execute it

### Current behavior:
```
$ ./Subsurface-4.6.4-x86_64.AppImage 
Started in /home/valentine/scratch ...
Moving to /tmp/.mount_AQx6Vb ...
Executing '/tmp/.mount_AQx6Vb/usr/bin/subsurface.wrapper' ...
counting args in %1$s/usr/lib/:%1$s/usr/lib/i386-linux-gnu/:%1$s/usr/lib/x86_64-linux-gnu/:%1$s/usr/lib32/:%1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/usr/lib/i386-linux-gnu/:%1$s/usr/lib/x86_64-linux-gnu/:%1$s/usr/lib32/:%1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/usr/lib/x86_64-linux-gnu/:%1$s/usr/lib32/:%1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/usr/lib32/:%1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/usr/lib64/:%1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib/:%1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib/i386-linux-gnu/:%1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib/x86_64-linux-gnu/:%1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib32/:%1$s/lib64/:%2$s
counting args in %1$s/lib64/:%2$s
  using LD_LIBRARY_PATH=/tmp/.mount_AQx6Vb/usr/lib/:/tmp/.mount_AQx6Vb/usr/lib/i386-linux-gnu/:/tmp/.mount_AQx6Vb/usr/lib/x86_64-linux-gnu/:/tmp/.mount_AQx6Vb/usr/lib32/:/tmp/.mount_AQx6Vb/usr/lib64/:/tmp/.mount_AQx6Vb/lib/:/tmp/.mount_AQx6Vb/lib/i386-linux-gnu/:/tmp/.mount_AQx6Vb/lib/x86_64-linux-gnu/:/tmp/.mount_AQx6Vb/lib32/:/tmp/.mount_AQx6Vb/lib64/:
  using PATH=./bin/:./sbin/:./games/:../bin/:../sbin/:/home/valentine/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/sbin:/sbin:/bin:/opt/texlive/2018/bin/x86_64-linux
  using PYTHONPATH=./share/pyshared/:
  using XDG_DATA_DIRS=./share/:
counting args in %1$s/usr/lib/qt4/plugins/:%1$s/usr/lib/i386-linux-gnu/qt4/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt4/plugins/:%1$s/usr/lib32/qt4/plugins/:%1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/i386-linux-gnu/qt4/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt4/plugins/:%1$s/usr/lib32/qt4/plugins/:%1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/x86_64-linux-gnu/qt4/plugins/:%1$s/usr/lib32/qt4/plugins/:%1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib32/qt4/plugins/:%1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib64/qt4/plugins/:%1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/qt5/plugins/:%1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/i386-linux-gnu/qt5/plugins/:%1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib/x86_64-linux-gnu/qt5/plugins/:%1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib32/qt5/plugins/:%1$s/usr/lib64/qt5/plugins/:%2$s
counting args in %1$s/usr/lib64/qt5/plugins/:%2$s
  using QT_PLUGIN_PATH=/tmp/.mount_AQx6Vb/usr/lib/qt4/plugins/:/tmp/.mount_AQx6Vb/usr/lib/i386-linux-gnu/qt4/plugins/:/tmp/.mount_AQx6Vb/usr/lib/x86_64-linux-gnu/qt4/plugins/:/tmp/.mount_AQx6Vb/usr/lib32/qt4/plugins/:/tmp/.mount_AQx6Vb/usr/lib64/qt4/plugins/:/tmp/.mount_AQx6Vb/usr/lib/qt5/plugins/:/tmp/.mount_AQx6Vb/usr/lib/i386-linux-gnu/qt5/plugins/:/tmp/.mount_AQx6Vb/usr/lib/x86_64-linux-gnu/qt5/plugins/:/tmp/.mount_AQx6Vb/usr/lib32/qt5/plugins/:/tmp/.mount_AQx6Vb/usr/lib64/qt5/plugins/:
  using PERLLIB=./share/perl5/:./lib/perl5/:
  using GSETTINGS_SCHEMA_DIR=./share/glib-2.0/schemas/:
/tmp/.mount_AQx6Vb/usr/bin/subsurface.wrapper: line 121:  8352 Aborted                 LD_LIBRARY_PATH="" zenity --question --title="$TITLE" --text="$TEXT" 2> /dev/null
/tmp/.mount_AQx6Vb/usr/bin/subsurface.wrapper: line 121:  8353 Aborted                 LD_LIBRARY_PATH="" zenity --question --title="$TITLE" --text="$TEXT" 2> /dev/null
/tmp/.mount_AQx6Vb/usr/bin/subsurface: error while loading shared libraries: libidn.so.11: cannot open shared object file: No such file or directory
```

### Mentions:

@probonopd: downloaded from the frontpage of https://appimage.org

probonopd, 2019-02-10:
Does this also happen with the AppImage from https://subsurface-divelog.org/download/?

Veyrdite, 2019-02-23:
Latest version from subsurf website:
```
$ ./Subsurface-4.8.5-x86_64.AppImage
./Subsurface-4.8.5-x86_64.AppImage: error while loading shared libraries: libkeyutils.so.1: cannot open shared object file: No such file or directory
```

If I install libkeyutils using my package manager then the appimage works and the program launches.  

Looks like the requirement for libidn has gone, but a new one for libkeyutils has appeared (earlier in the exec process).

buxtonpaul, 2019-09-26:
So looking into this. It looks like libkeyutils.so is one of the packages explicitly excluded by the Appimage build scripts, as Appimage expects it to be on most systems.
https://github.com/probonopd/linuxdeployqt/blob/master/tools/linuxdeployqt/excludelist.h

Quickly checking a Fedora vm, the library does appear to be present so it is not just a Debian thing. Not sure if this is something the Appimage guys will want to change or not. Perhaps probonopd can comment as he appears to be heavily involved in appimage stuff...

dirkhh, 2019-09-26:
@probonopd can you help?
And for @buxtonpaul - Simon is not "heavily involved". He's the person who came up with the idea and the main developer behind AppImage. This is like saying that Linus is "heavily involved" in Linux (actually, it's even less accurate than that because the percentage of the AppImage work done by Simon is orders of magnitude higher than Linus' percentage of Linux work)

buxtonpaul, 2019-09-26:
Ah! I hope I havent caused offence!
The linuxdeploqt.appimage doesn't come from what looks like the main
Appimage repository. So I didn't realise it was by the same person. It is a
fantastic tool!

Regarding actually being able to do something about this, I am not clear
what the best course is. Given that the library in question is pretty
standard on several distributions, and has in the past been the target of a
rootkit, is it something we should be including in the package? It does
look to be available as a package from Voidlinux, so would we be better
asking users to install it?

Paul

probonopd, 2019-09-26:
So far we simply had assumed that `libkeyutils.so` comes as part of the default installation of all desktop-oriented distributions, but apparently you found an exception. It is only 14k in size and has no uncommon dependencies, so I don't see a reason why it could not be bundled, unless I am overlooking something. I guess we need to try to find out.

glance-, 2019-11-10:
libkeyutils.so.1 is now shipped in the app image.

closed 2019-11-10 by glance-

zdavatz, 2020-03-05:
Not really. I still got the same error: 
```
/tmp> ./Subsurface-4.9.3-x86_64.AppImage 
./Subsurface-4.9.3-x86_64.AppImage: error while loading shared libraries: libkeyutils.so.1: cannot open shared object file: No such file or directory
```
then I installed `libkeyutils` manually and now it works.