subsurface/subsurface #1972: Many links not working

holmesmi, 2019-02-10:
### Describe the issue:

- X Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

MacOS

### Subsurface version:

### Steps to reproduce:

Attempting to download the mac version from the website, or go to the video tutorial, results in links that go nowhere.

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-02-10:
The tutorial video is so old that I removed it. Need to fix that the links are still there (I thought I did that already. Oops)
The Mac download is more concerning. I'll look at it.
Thanks

dirkhh, 2019-02-11:
The download links should be fixed. that was an odd error. I still need to remove the links to the tutorial video

gerardsnaauw, 2019-03-26:
Downloads are still not working properly.
Depending on what language the page is using, the download refer to either subsurface-divelog.org/download/file or subsurface-divelog.org/download**s**/file.
Only the download**s** one works.

dirkhh, 2019-03-28:
I believe this has finally been fixed. Thanks for pointing out this oversight

fmatray, 2019-08-02:
Somes links are still broken in the french version :
https://subsurface-divelog.org/fr/documentation/

mikeller, 2026-10-17:
Hi @holmesmi,

Good news, this is resolved. The website has since been rebuilt from scratch, and I've just directly checked every broken link described in this thread: the French documentation page (subsurface-divelog.org/fr/documentation/) now loads correctly, the current-release page's Windows and macOS download links both work, and there's no remaining reference to the old tutorial video anywhere on the site.

Closing this as fixed by the subsequent site rebuild. Please reopen with the specific broken link(s) if you find something still pointing nowhere on the current site.

closed 2026-10-17 by mikeller