subsurface/subsurface #1975: Changing dive time/ date behaviour

cmarlowe, 2019-02-17:
[ ] Bug

### Issue long description:
I've just updated to latest Mac version 4.8.5. 

I have a number of different models of computers, and have been aligning date and time for some dives where the times weren't aligned.

Previously I was able to change date and time in the Notes tab successfully. Since update to latest version, if I try to amend the time or date, when I select 'accept changes' the time and date in the dive list changes to a date time which bear no resemblance to the date / time changed to. Initially the date and time display as the correct revised date time in the notes page, but if I click into another dive and back in, the notes page also changes to the incorrect date time. 

As an example, I have tried several times to make a time change (which was initially supposed to be a change of an hour).  Each time it makes it worse. e.g. current dive date time in Notes tab is Thu 21 Feb 2019 22.28, on trying to change it to Fri 15th Feb 11:00 it changes to Thurs 28 Feb 2019 9:56. If I try again to set it to 15th Feb at 11:00 it now changes to Web 13th Marc 2019 8:52... (and so on). 

### Operating system:

Mac Mojave 10.14.2

Multiple dive computers. This example happens to be a Perdix, but the same applies to multiple models. 

### Subsurface version:

Latest Mac release 4.8.5 

official release

### Steps to reproduce:

1) Upload a dive 
2) Enter notes tab
3) Change the time
4) Accept changes
5) See date/time in the dive list. 

### Current behavior:

As above

### Expected behavior:

Date time should change correctly. 

### Additional information:

### Mentions:

mturkia, 2019-02-17:
I have noticed the same problem. A workaround is to select dives that need time adjustment and choose "shift dive times" from the right click menu.

cmarlowe, 2019-02-18:
Thanks mturkia, I'll try that in the short term

bstoeger, 2019-02-18:
Certainly due to divelist-undo. I will look into it.

closed 2019-02-18 by dirkhh

ssoloway, 2019-02-20:
I have noticed the same problem when changing the date, and adjusting the time by multiples of 24 doesn't seem to solve the problem.  Is there a workaround?

dirkhh, 2019-02-20:
Yes, we have test builds of the next version which fixes this issue.
Which OS do you need - happy to point you to such a binary

cmarlowe, 2019-02-21:
I'm on macOS Mojave 10.14.2, a test build would be fab, thanks

dirkhh, 2019-02-21:
Try this one: https://subsurface-divelog.org/downloads/test/Subsurface-4.8.4-253-g5033c02a110e.dmg
(yes, the version number is 4.8.4-something... it's a long story, but yes, this is indeed newer than 4.8.5)

cmarlowe, 2019-02-22:
I replaced my version of Subsurface with this one and now get an error message saying Subsurface cannot be opened because of a problem:  “Check with the developer to make sure Subsurface works with this version of macOS. You may need to reinstall the application. Be sure to install any available updates for the application and macOS.”

cmarlowe, 2019-03-29:
Hi, I wondered if there was an estimated release date for the next version including this amend? I'm unable to get the latest test build version to open at all, so it would be great to have a rough idea.  Many thanks, Celia

dirkhh, 2019-03-29:
We don't have a schedule, yet.
But I missed that you were unable to open the test build.
Which version of macOS are you running? On what kind of Mac?

cmarlowe, 2019-03-29:
Hi Dirk,

Thanks for getting back to me.

I’m on a Macbook Air, running Mojave 10.14.3.

Thanks,

Celia

atdotde, 2019-03-29:
Dirk,

> On 29. Mar 2019, at 14:54, Dirk Hohndel <notifications@github.com> wrote:
> 
> We don't have a schedule, yet.
> But I missed that you were unable to open the test build.
> Which version of macOS are you running? On what kind of Mac?
> 
> 

I just tried to run the file myself. I crashes for me as well complaining about a missing jpeg library:

dyld: Library not loaded: /usr/local/opt/jpeg-turbo/lib/libjpeg.8.dylib

Best
Robert

dirkhh, 2019-03-29:
Thanks, Robert. It works on my system so I'm guessing that I have that library and therefore don't notice that it's missing in the bundle.
I'll try to create a new bundle today and will post here once I have that uploaded.

dirkhh, 2019-03-29:
Please give this one a try. It should be completely self-contained...
https://subsurface-divelog.org/downloads/test/Subsurface-4.8.4-347-g30746e5d3f6b.dmg

cmarlowe, 2019-03-30:
Hi Dirk,

That opens fine and I can now change dates successfully, thanks!

There is one strange thing different to usual in that it seems to be fixed in height to the top and bottom of the screen (I can change the width) so I can’t see the bottom dives unless I reorder,  as they are overlaid by the dock. It’s not a biggie, I can work around it, but just FYI.

Many thanks for sending this Dirk, my life will be a whole lot easier now!

Celia

https://subsurface-divelog.org/downloads/test/Subsurface-4.8.4-347-g30746e5d3f6b.dmg<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fsubsurface-divelog.org%2Fdownloads%2Ftest%2FSubsurface-4.8.4-347-g30746e5d3f6b.dmg&data=02%7C01%7Cc.marlowe%40uea.ac.uk%7C487e2f59c0fb401aae0608d6b46717a3%7Cc65f8795ba3d43518a070865e5d8f090%7C0%7C0%7C636894752200900870&sdata=eFCqw3iG3CTS%2B%2BLcTEhP%2B%2BCaavM3RvRlS1qJYV1XvoQ%3D&reserved=0>

PR #1976 (2019-02-18): Edit: use correct offset when changing dive times
The undo-work reversed the direction of the offset. This was apparently
only fixed when using the menu entry, but not when editing dives directly.
Invert the offset to get the correct time.

While doing so, remove a redundant if: First it checked whether the
dates are the same, then whether the offset is non-zero.

Fixes #1975.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long descripti [... cut]
commit 68b5493c (2019-02-18): Edit: use correct offset when changing dive times

The undo-work reversed the direction of the offset. This was apparently
only fixed when using the menu entry, but not when editing dives directly.
Invert the offset to get the correct time.

While doing so, remove a redundant if: First it checked whether the
dates are the same, then whether the offset is non-zero.

Fixes #1975.
commit a3a1a74d (2019-04-04): Edit: use correct offset when changing dive times

The undo-work reversed the direction of the offset. This was apparently
only fixed when using the menu entry, but not when editing dives directly.
Invert the offset to get the correct time.

While doing so, remove a redundant if: First it checked whether the
dates are the same, then whether the offset is non-zero.

Fixes #1975.