subsurface/subsurface #1983: Subsurface-mobile: Swiftkey, word removed when hitting Enter

plxflx, 2019-02-21:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When writing log notes the current/last word is removed when hitting Enter to get a new line.

### Operating system:
Android 8.0.0 on Samsung device

### Subsurface version:
Subsurface-mobile 2.1.6 (4.8.4.66)
Swiftkey 7.2.3.24
When using Samsung keyboard this problem does not appear. This makes me think Swiftkey will be blamed here but let me say that I don't have this problem in any other part of my phone or any other app. 

### Steps to reproduce:

1) Write dive notes "Best dive ever!"
2) Press Enter on keyboard

### Current behavior:

Dive note contain "Best dive ever!"
After pressing Enter with marker at end of line it looks like this:
"Best dive
"

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-02-21:
Hrmpf. That sucks.
This is of course something that happens inside libraries that we use - so this is outside of our control. Most likely this is a Qt/QML issue. I'll do some googling to figure out if this is a known issue, but I don't have a lot of hope that there's a known workaround.

dirkhh, 2019-02-21:
OK, this is a known bug: https://bugreports.qt.io/browse/QTBUG-63653
And no, there's no fix, no workaround at this point.

plxflx, 2019-02-21:
More then a year old bug. Small hop of fix then :-( 
Might be time to change keyboard. Thank you for your help!

dirkhh, 2019-02-21:
Actually, that's not necessarily how the Qt team works. The fact that there's no discussion on the bug is indeed a concern, but it often takes a while for things like this to bubble up higher in Priority. Feel free to add to the bug with your own report :-)

plxflx, 2019-02-21:
I will give it a try. Those who can't code, report bugs?

mikeller, 2026-10-17:
Hi @plxflx,

This traces to a known, long-standing upstream Qt bug (https://bugreports.qt.io/browse/QTBUG-63653) affecting third-party keyboards like Swiftkey on Android, not something in Subsurface's own code, consistent with it not reproducing on the stock Samsung keyboard.

Closing this citing that upstream bug as the root cause. Switching keyboards (as you found) is the practical workaround until/unless Qt fixes it upstream. Please reopen if you find this reproduces even with a non-Swiftkey keyboard.

closed 2026-10-17 by mikeller