subsurface/subsurface #2029: The AppImage does not work on Debian Buster with libSSL issue

escafandra, 2019-03-30:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The subsurface AppImage version 4.8.5 does not run on Debian 10.0 producing this error related to libSSL:
`error:25066067:DSO support routines:DLFCN_LOAD:could not load the shared library:dso_dlfcn.c:185:filename(libssl_conf.so): libssl_conf.so: cannot open shared object file: No such file or directory
140104002685376:error:25070067:DSO support routines:DSO_load:could not load the shared library:dso_lib.c:244:
140104002685376:error:0E07506E:configuration file routines:MODULE_LOAD_DSO:error loading dso:conf_mod.c:285:module=ssl_conf, path=ssl_conf`
### Operating system:

Debian Buster Alpha 5 Release
libssl 1.1 installed
libcurl4 installed
### Subsurface version:

Subsurface AppImage 4.8.5
### Steps to reproduce:

Just run the AppImage in Debian Buster

### Mentions:

bcogel, 2019-04-29:
Same here. Workaround is to create a symlink:

ln -s  /usr/lib/x86_64-linux-gnu/libssl.so.1.1 /usr/lib/x86_64-linux-gnu/libssl_conf.so

bcogel, 2019-04-29:
But.....

Because I wanted to report another bug, I called Subsurface with LANG=C .... and it crashed....

berthold@tux:~$ LANG=C Subsurface-4.8.5-x86_64.AppImage 
Auto configuration failed
139633179007424:error:2506406A:DSO support routines:DLFCN_BIND_FUNC:could not bind to the requested symbol name:dso_dlfcn.c:287:symname(OPENSSL_finish): /usr/lib/x86_64-linux-gnu/libssl_conf.so: undefined symbol: OPENSSL_finish
139633179007424:error:2506C06A:DSO support routines:DSO_bind_func:could not bind to the requested symbol name:dso_lib.c:294:
139633179007424:error:0E07606D:configuration file routines:MODULE_RUN:module initialization error:conf_mod.c:235:module=ssl_conf, value=ssl_sect, retcode=-1      
QMutex: destroying locked mutex
[1]+  Fertig                  Subsurface-4.8.5-x86_64.AppImage
Speicherzugriffsfehler

Calling again without LANG=C

berthold@tux:~$ Subsurface-4.8.5-x86_64.AppImage 
Auto configuration failed
139818649799104:error:2506406A:DSO support routines:DLFCN_BIND_FUNC:could not bind to the requested symbol name:dso_dlfcn.c:287:symname(OPENSSL_finish): /usr/lib/x86_64-linux-gnu/libssl_conf.so: undefined symbol: OPENSSL_finish
139818649799104:error:2506C06A:DSO support routines:DSO_bind_func:could not bind to the requested symbol name:dso_lib.c:294:
139818649799104:error:0E07606D:configuration file routines:MODULE_RUN:module initialization error:conf_mod.c:235:module=ssl_conf, value=ssl_sect, retcode=-1      
QMutex: destroying locked mutex
Speicherzugriffsfehler

And now calling with LANG=de_DE Subsurface-4.8.5-x86_64.AppImage .....

I got it working again..... Strange....

escafandra, 2019-05-08:
> Same here. Workaround is to create a symlink:
> 
> ln -s /usr/lib/x86_64-linux-gnu/libssl.so.1.1 /usr/lib/x86_64-linux-gnu/libssl_conf.so

Thanks @bcogel , the workaround worked. I faced also the next issue you mentioned later. It happened even calling subsurface without parameters. It also worked for me using  `LANG=de_DE`

hhassey, 2019-05-17:
Have the same issue on Debian Buster and Subsurface App image 4.8.6 Hope this can be fixed soon

dirkhh, 2019-05-17:
I don't think anyone is working on this right now - so unless someone sends patches, I don't see it getting fixed any time soon

hhassey, 2019-05-29:
Is this a Debian or a Subsurface issue?

hhassey, 2019-05-31:
This seems to be ralated to #1740

glance-, 2019-11-10:
Openssl issue fixed via #2296

closed 2019-11-10 by glance-