subsurface/subsurface #2032: Date change shifts in wrong direction

RealNagus, 2019-03-31:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I just downloaded the latest 4.8.5 release and imported some new dives from my Mares Nemo Wide 2 computer. For some of the logs the date is several years off because I forgot the set the clock after replacing the battery. When I try to correct the date and time in the edit view it does change the date, but in the wrong direction. E.g. a  dive that was logged as 2012 and corrected to 2019 appears as 2005. Additional edits move the entry further back in time.

This does not happen with the 4.8.4 release. 

### Operating system:

GNU/Linux, Debian Stretch, Laptop computer

### Subsurface version:

Subsurface 4.8.5 (AppImage), Official release

### Steps to reproduce:

[1. Import log entry with date in the past]
2. Edit the log entry's date, hit apply.
3. Date is shifted further into the past.

### Current behavior:

Date corrections move into the wrong direction. This also works the other way around, btw. Setting the date further into the past will move the date into the future. 

### Expected behavior:

Well, apply the actual date that was entered. ;-)

### Additional information:
A logfile with a single entry that shows the behavior is attached. The entry was logged as 2012. When I change it to 2019 it will be set as 2005. Editing it again will set it to 1998 and so on.

[datechange_issue_example.xml.gz](https://github.com/Subsurface-divelog/subsurface/files/3026581/datechange_issue_example.xml.gz)

### Mentions:

bstoeger, 2019-03-31:
Thanks for your report - this is fixed in current master.

bstoeger, 2019-04-13:
Fixed in upcoming release.

closed 2019-04-13 by bstoeger