subsurface/subsurface #2037: Not able to parse CSV in 4.8.5

jofan, 2019-04-04:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

In the latest version I'm no longer able to import CSV files that worked fine in previous versions.
This is an example of rows in the CSV (#, location, date, max depth, duration, time, notes, buddy):

```
1202	LCU	2005-02-19	14	0:21:00	10:00:00	Fisheye photography	Johanna
1203	Makaha	2005-02-19	10	0:40:00	12:00:00	Intros for Captain Bruce	Johanna
```
I have tried with different date and time formats but not been able to make it work.

### Operating system:

macOS Mojave
English
MacBook Pro 2015

### Subsurface version:

4.8.5

I does not happen to earlier versions.

I'm using the official release

### Steps to reproduce:

1. Copy and paste log rows from Excel into Google Sheets
2. In Google Sheets: select "Download as Tab-separated values"
3. Rename downloaded file from *.tsv to *.csv
4. Select "Import log files" in Subsurface
5. Drag and drop header tags appropriately and import
6. Error message is displayed saying that the CSV could not be parsed.

### Current behavior:

Error message generated that the CSV cannot be parsed

### Expected behavior:

Parsing CSV and importing logs

### Additional information:

Example file attached (rename .txt to .csv)

### Mentions:

[Divelogexport - Sheet22.txt](https://github.com/Subsurface-divelog/subsurface/files/3042302/Divelogexport.-.Sheet22.txt)

jofan, 2019-04-04:
Having tested some more it seems like it's specific to this file. I'll try to find the row that causes the issue.

atdotde, 2019-04-04:
Possibly any line with quotation marks or an apostrophe?

jofan, 2019-04-04:
Thanks @atdotde, it was a & sign that caused the issue.

atdotde, 2019-04-04:
Ampersand should be a legal character in CSV. This is a bug.

mturkia, 2019-07-13:
As we are using XSLT to parse CSV, it looks like this cannot be fixed without re-writing the CSV import in C or C++.

dirkhh, 2019-07-13:
Can this be handled by a pre-parsing step? Where we quote special characters that XSLT can't handle?

commit 90b0e38e (2020-02-06): CSV import: fix importing ampersand character

As we do XSLT parsing for the CSV import, ampersand characters need to
be encoded with &amp; for the parsing to succeed.

Fixes #2037
closed 2020-02-06 by dirkhh