subsurface/subsurface #2045: Dive site filtering on a site with no dives is broken

djunkins, 2019-04-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When in the dive site tab, if you select a dive site that has zero dives associated with it, the dive list is empty, but both the main window tabs and the dive site table are grayed out which does not allow the user to unselect the site to reset the filter. The same problem exists when you use the `+` button to add a dive site to the table since the moment the dive site is created it has no dives associated with it.

The only workaround I found was to use <command-F> to bring up the filter pane and reset the filter there.
### Operating system:

MacOS 10.14.4

### Subsurface version:

This is on the recently merged `master 8695d8bdb` branch with the dive site undo features.

### Steps to reproduce:

1) Navigate to `Dive sites` tab
2) Use `+` button to create a dive site in the dive site table
3) Struggle to find out how to edit or unselect the dive site or reset the filter

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

djunkins, 2019-04-12:
I also just noticed and verified that after triggering the bug above, MacOS gives an error saying Subsurface closed unexpectedly when the user exits from the program through the normal process (<Command-Q>).

dirkhh, 2019-04-12:
That might be the one I'm describing in #2046

djunkins, 2019-04-12:
Yes, the crash on exit sounds the same. I did not attempt to isolate it to filtering on an unused dive site vs. filtering on any dive site.

bstoeger, 2019-04-12:
That shouldn't be to hard to fix. But while doing so, I found a cognate bug: On File>Close, the dive site table keeps its entries and adding a new dive site will add it at the end.

bstoeger, 2019-04-13:
Should be fixed in master.

closed 2019-04-13 by bstoeger