subsurface/subsurface #2046: Crash on exit when filtering dive list by selecting a dive site

dirkhh, 2019-04-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

This is easy to reproduce.
Pick a dive site in the dive site tab - the dive list is now filtered to only show dives at that site.
Now quit Subsurface.
Crash

### Operating system:

Tested on Linux

### Subsurface version:

latest master 

### Additional information:

stack trace seems to indicate that in the `MainWindow` destructor we end up calling down to `MultiFilterSortModel::myInvalidate()` which emits `filterFinished()` which then calls `mainTab->updateDiveInfo()` which dereferences `MainWindow::instance()->graphics->isPlanner()` and crashes.

### Mentions:

@bstoeger

bstoeger, 2019-04-12:
> This is easy to reproduce.

Doesn't happen for me... :)

But we had that bug before when the filter was active... I'll try to remember what that was.

bstoeger, 2019-04-12:
Oh yes, now I remember: somehow some Qt version changed and would call hide() on child objects during destruction(!). I was pretty upset back then, because a destructor should free memory and *not* call methods that might invoke things in a partially destructed object tree. If this is the same thing, I consider it a bug in Qt. Evidently, we still have to work around it. The last time the solution was transforming child-objects into sub-objects, I believe.

Might also be something completely different, of course...

Could you post the backtrace?

dirkhh, 2019-04-12:
What I list above is the back trace - with a few Qt internal functions in the middle. But those are all of our functions that we call.

bstoeger, 2019-04-13:
> What I list above is the back trace - with a few Qt internal functions in the middle. 

Those functions and objects would be important, because they explain why it doesn't happen for me.

bstoeger, 2019-04-13:
I think I understand what's happening:

New Qt-versions call hide on child-objects in the destructor. If the dive-site tab is open, this calls hide, which resets the filter, which accesses the semi-destructed windows object.

I believe this to be a Qt bug: It seems quite normal that child objects would call into their parent on hiding. Calling hide() in the destructor of the *base* class violates what you would expect from a C++ destructor. You're supposed to free resources and nothing else. It seems to me that the whole Qt-object system reverses logical order of destruction: the child objects should be destroyed *first*.

Perhaps we simply should make the maintab a smart-pointer. That will destroy the maintab before the mainwindow and remove it from Qt's object hierarchy.

bstoeger, 2019-04-13:
Should be fixed in master.

closed 2019-04-13 by bstoeger