subsurface/subsurface #2065: When i open app again then is a Black screen and must be closed

alex-1987, 2019-04-21:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

If i open the app and go back to home screen and want to open the surface mobile again i have Black screen and i must close the app. 

### Operating system:

Android 8.0.0

Lg g7

### Subsurface version:

2.1.6(4.8.4.66)

Official app

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

alex-1987, 2019-04-21:
---------- subsurface.log ----------
Empty filename passed to function
"0.004: Failed to open logfile /storage/emulated/0/subsurface.log at So. Apr. 21 17:10:42 2019 error: Permission denied"
"0.004: Failed to open logfile /storage/emulated/0/Documents/subsurface.log at So. Apr. 21 17:10:42 2019 error: No such file or directory"
"0.004: Successfully opened logfile /storage/emulated/0/Android/data/org.subsurfacedivelog.mobile/files/Documents/subsurface.log at So. Apr. 21 17:10:42 2019"
"0.004: Starting Subsurface-mobile:2.1.6(4.8.4.66):Android Oreo (8.0):arm:de-DE"
"0.004: built with libdivecomputer v0.7.0-devel-Subsurface-NG (072bef16666e0ec397b8236e92722b2f1347b59b)"
"0.004: built with Qt Version 5.12.0, runtime from Qt Version 5.12.0"
"0.004: built with libgit2 0.26.0"
BTDiscoveryReDiscover: localBtDevice.isValid() true
"BTDiscoveryReDiscover: localDevice alex-G7 ThinQ is powered on, starting discovery"
paired BT classic device type 1 with address "00:1E:7C:62:FB:FD"
paired BT classic device type 1 with address "00:0D:18:A1:12:74"
paired BT classic device type 1 with address "00:12:1C:48:F8:19"
paired BLE device type 2 with address "LE:0C:8C:DC:06:43:85"
paired BT classic device type 1 with address "00:54:AF:0A:31:24"
Found new device: "JBL Endurance JUMP" "00:1E:7C:62:FB:FD"
Not recognized as dive computer
Found new device: "Alex-Car" "00:0D:18:A1:12:74"
Not recognized as dive computer
Found new device: "SUBARU v 5.08" "00:12:1C:48:F8:19"
Not recognized as dive computer
Found new device: "EON Steel" "LE:0C:8C:DC:06:43:85"
"this could be a Suunto EON Steel"
Found new device: "My_Radio" "00:54:AF:0A:31:24"
Not recognized as dive computer
Paired = "JBL Endurance JUMP" "00:1E:7C:62:FB:FD"
Paired = "Alex-Car" "00:0D:18:A1:12:74"
Paired = "SUBARU v 5.08" "00:12:1C:48:F8:19"
Paired = "EON Steel 1514100268" "LE:0C:8C:DC:06:43:85"
Paired = "My_Radio" "00:54:AF:0A:31:24"
"Created position source android"
"0.017: Created position source android"
"Set GPS service update interval to 300 s"
"0.017: Set GPS service update interval to 300 s"
"0.018: location service is available"
qrc:/org/kde/kirigami/Page.qml:276:5: QML Loader: Binding loop detected for property "active"
Model size of -1 is less than 0
"0.422: Datendatei wird synchronisiert"
"0.424: Lade Tauchgänge aus dem lokalen Zwischenspeicher"
"0.567: Tauchdaten erfolgreich geöffnet"
"0.587: AppState changed to active with save ongoing and no unsaved changes"
"0.590: 313 dives loaded from cache"
"0.591: have cloud credentials, but user asked not to connect to network"
checkPendingIntents 
Using the following font: Roboto at 17pt with mobile_scale: 1
qqwindow devicePixelRatio 3.50271 3.50271
Supported dive computers:
"Aeris: 500 AI (SERIAL), A300 (SERIAL), A300 AI (SERIAL), A300CS (SERIAL), Atmos 2 (SERIAL), Atmos AI (SERIAL), Atmos AI 2 (SERIAL), Compumask (SERIAL), Elite (SERIAL), Elite T3 (SERIAL), Epic (SERIAL), F10 (SERIAL), F11 (SERIAL), Manta (SERIAL), XR-1 NX (SERIAL), XR-2 (SERIAL)"
"Aqualung: i200 (SERIAL), i300 (SERIAL), i300C (SERIAL, BLE), i450T (SERIAL), i550 (SERIAL), i750TC (SERIAL, BT), i770R (SERIAL, BLE)"
"Atomic Aquatics: Cobalt (USB), Cobalt 2 (USB)"
"Beuchat: Mundial 2 (SERIAL), Mundial 3 (SERIAL), Voyager 2G (SERIAL)"
"Cochran: Commander I (SERIAL), Commander II (SERIAL), Commander TM (SERIAL), EMC-14 (SERIAL), EMC-16 (SERIAL), EMC-20H (SERIAL)"
"Cressi: Drake (SERIAL), Giotto (SERIAL), Leonardo (SERIAL), Newton (SERIAL)"
"Garmin: Descent Mk1 (USBSTORAGE)"
"Genesis: React Pro (SERIAL), React Pro White (SERIAL)"
"Heinrichs Weikamp: Frog (SERIAL, BT), OSTC (SERIAL), OSTC 2 (SERIAL, BT, BLE), OSTC 2 TR (SERIAL, BT, BLE), OSTC 2C (SERIAL), OSTC 2N (SERIAL), OSTC 3 (SERIAL), OSTC 4 (SERIAL, BT, BLE), OSTC Mk2 (SERIAL), OSTC Plus (SERIAL, BT, BLE), OSTC Sport (SERIAL, BT, BLE), OSTC cR (SERIAL)"
"Hollis: DG02 (SERIAL), DG03 (SERIAL), TX1 (SERIAL)"
"Mares: Puck Pro (SERIAL, BLE), Quad (SERIAL, BLE), Quad Air (SERIAL, BLE), Smart (SERIAL, BLE), Smart Air (SERIAL, BLE)"
"Oceanic: Atom 1.0 (SERIAL), Atom 2.0 (SERIAL), Atom 3.0 (SERIAL), Atom 3.1 (SERIAL), Datamask (SERIAL), F10 (SERIAL), F11 (SERIAL), Geo (SERIAL), Geo 2.0 (SERIAL), OC1 (SERIAL), OCS (SERIAL), OCi (SERIAL), Pro Plus 2 (SERIAL), Pro Plus 2.1 (SERIAL), Pro Plus 3 (SERIAL), Pro Plus X (SERIAL, BLE), VT 4.1 (SERIAL), VT Pro (SERIAL), VT3 (SERIAL), VT4 (SERIAL), VTX (SERIAL), Veo 1.0 (SERIAL), Veo 180 (SERIAL), Veo 2.0 (SERIAL), Veo 200 (SERIAL), Veo 250 (SERIAL), Veo 3.0 (SERIAL), Versa Pro (SERIAL)"
"Scubapro: Aladin Sport Matrix (BLE), Aladin Square (USBHID), G2 (USBHID, BLE), G2 Console (USBHID, BLE)"
"Seemann: XP5 (SERIAL)"
"Shearwater: Nerd (SERIAL, BT), Nerd 2 (BLE), Perdix (SERIAL, BT, BLE), Perdix AI (BLE), Petrel (SERIAL, BT), Petrel 2 (SERIAL, BT, BLE), Predator (SERIAL, BT), Teric (BLE)"
"Sherwood: Amphos (SERIAL), Amphos Air (SERIAL), Insight (SERIAL), Insight 2 (SERIAL), Vision (SERIAL), Wisdom (SERIAL), Wisdom 2 (SERIAL), Wisdom 3 (SERIAL)"
"Subgear: XP-Air (SERIAL)"
"Suunto: Cobra (SERIAL), Cobra 2 (SERIAL), Cobra 3 (SERIAL), D3 (SERIAL), D4 (SERIAL), D4f (SERIAL), D4i (SERIAL), D6 (SERIAL), D6i (SERIAL), D9 (SERIAL), D9tx (SERIAL), DX (SERIAL), EON Core (USBHID, BLE), EON Steel (USBHID, BLE), Eon (SERIAL), Gekko (SERIAL), HelO2 (SERIAL), Mosquito (SERIAL), Solution (SERIAL), Solution Alpha (SERIAL), Solution Nitrox (SERIAL), Spyder (SERIAL), Stinger (SERIAL), Vyper (SERIAL), Vyper 2 (SERIAL), Vyper Air (SERIAL), Vyper Novo (SERIAL), Vytec (SERIAL), Zoop (SERIAL),  [... cut ...]
"Uwatec: Aladin Air Twin (SERIAL), Aladin Air Z (SERIAL), Aladin Air Z Nitrox (SERIAL), Aladin Air Z O2 (SERIAL), Aladin Pro (SERIAL), Aladin Pro Ultra (SERIAL), Aladin Sport Plus (SERIAL), Memomouse (SERIAL)"
qqwindow screen has ldpi/pdpi 71.9442 161.41
qrc:/org/kde/kirigami/GlobalDrawer.qml:258: TypeError: Cannot read property 'length' of undefined
qrc:/org/kde/kirigami/GlobalDrawer.qml:293: TypeError: Cannot read property 'length' of undefined
qrc:/org/kde/kirigami/GlobalDrawer.qml:292: TypeError: Cannot read property 'length' of undefined
---------- finish ----------

mikeller, 2026-10-17:
Hi @alex-1987,

This 2019 report predates several subsequent reworks of the mobile app's lifecycle/UI handling, which is the most likely explanation for a black screen on reopen. With no further detail and no corroborating reports since, we're closing this as stale.

Please reopen with your current Android/Subsurface-mobile version if you're still seeing a black screen when reopening the app.

closed 2026-10-17 by mikeller