subsurface/subsurface #2066: Changes lost when saving to cloud

kittykat, 2019-04-22:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I imported some dives and was editing the description. After applying the changes, saved the file (I use the cloud storage) but my changes disappeared and my previous save was reloaded. The only thing that I did differently from usual is that I was writing the log slowly. Same thing happened this morning and I lost my description again. No useful information in the terminal (was running Subsurface from a terminal to debug new computer issues). I did not have Subsurface opened on any other device at the time.

I have had issues last two weeks with connecting to the cloud storage for saving my log, but not in the last few days when I've been seeing data loss.

I think the issue might be tied to leaving Subsurface running all the time as subsequent saves immediately after the failed one work fine.

### Operating system:
Linux desktop

### Subsurface version:
4.8.6

### Steps to reproduce:
1. Import dive and save.
1. Edit the description slowly (maybe over an hour or so).
1. Apply Changes.
1. Press Ctrl+S to save.
1. See changes vanish after cloud sync is complete.

### Current behavior:
Sometimes the changes are lost.

### Expected behavior:
The changes are pushed to the server.

### Additional information:

### Mentions:

bstoeger, 2019-04-26:
Thank you for the report. The whole dive-editing was refactored pretty radically recently. Would it be possible for you to try current master?

kittykat, 2019-04-29:
I'll give it a go tomorrow

dirkhh, 2019-05-07:
Any feedback?

janiversen, 2019-06-02:
master - 2days on a Mac, still have this problem. I e.g. edited weight on a number of dives, saved to cloud, restarted (and thereby reloaded), all changes gone.

My current workaround is to put something in the note field, that causes save to work correctly.

Be aware some fields work others not.

bstoeger, 2019-06-02:
Thanks for report - should be fixed by #2118.
(Though I think it's a different issue from the one reported by @kittykat)

kittykat, 2019-06-30:
Seems to be working ok with 4.8.4-663-g5c1446a87acb

bstoeger, 2019-06-30:
Thanks for confirmation - closing (for now).

closed 2019-06-30 by bstoeger