subsurface/subsurface #2069: Download from computer no more working in 4.8.6 on win10

jupiler, 2019-04-25:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Download via usb from DC no more working in 4.8.6
downgrade to 4.8.5 and it works again as expected.

### Operating system:
OS win 10 1803 build 17134.706 English
 DC scubapro G2 

### Subsurface version:
 4.8.6

 No

official release

### Steps to reproduce:

1) Connect DC to computer put in log mode, 
2) Open subsurface import from divecomputer
3) select computer and press download
At this point i get an error "Unable to open Scubapro(G2)"
and at the bottom a red banner with "Unsupported operation"

### Current behavior:

No downloads from DC

### Expected behavior:

Being able to download my dives from my DC

### Additional information:

Both logfiles seem to be empty 

### Mentions:

dirkhh, 2019-05-07:
There's a chance that a library is missing in the 4.8.6 installer. Those builds are automated and should do the right thing, but... yeah, that's what it looks like. I'll investigate

jupiler, 2019-05-07:
Downloaded yesterday the latest build. to test the dive sites export. Tested the download again but i still got the same error.

noseygit, 2019-05-07:
Im seeing a similar problem downloading from bluetooth - again on 4.8.6 whilst running W10.

4.8.5 works fine, 4.8.6 says that it cant open the bluetooth adapter.

jupiler, 2019-08-10:
4.9.0 still the same issue

jb2cool, 2020-03-08:
@jupiler @noseygit reading through your issue i think this may be an easy thing to fix for you. You previously said your steps were:

1. Connect DC to computer put in log mode,
2. Open Subsurface import from divecomputer
3. Select computer and press download
4. At this point i get an error "Unable to open Scubapro(G2)"
5. and at the bottom a red banner with "Unsupported operation"

Please can you try doing something slightly different for me. When you select your Vendor and Dive computer i think we really need to also populate the Device or mount point field, this is usually done by checking the box at the bottom that says "Choose Bluetooth download mode"

The steps I'd like you to try would be:

1. Put DC into log mode,
2. Open Subsurface > Import > Import from divecomputer
3. Select computer and chech the "Choose Bluetooth download mode"
4. In the box that pops up, click on your computer and click save
5. Press the download button

It would be worth updating to the latest version (4.9.3) before you try this so that you are on the most up to date release.

mikeller, 2026-10-17:
Hi @jupiler,

This report is from Subsurface 4.8.6 on Win10 in 2019, download handling has changed substantially since then, and we've had no corroborating reports or follow-up. We're closing this as stale.

Please reopen with your current version and a verbose log if downloading from your dive computer still fails.

closed 2026-10-17 by mikeller