subsurface/subsurface #2076: Can not edit dive site location on map

djunkins, 2019-04-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When editing a dive site from the dive site tab, I can no longer drag the flag on the map to update the lat/long of the dive site location.

### Operating system:

OSX 10.14.4
### Subsurface version:

* master          9bb7bbef8 [origin/master] Support for Scubapro G2 HUD discovery via bluetooth

### Steps to reproduce:

1) Navigate to dive site tab
2) Select the site to be edited
3) Enter edit mode
4) Attempt to drag flag on the site
### Current behavior:

### Expected behavior:

### Additional information:

It looks like the grayed out flag for sites that do not have any dives associated with them is missing and the map is not moving to the location of those sites when you attempt to edit them as well.

### Mentions:

@bstoeger

bstoeger, 2019-04-28:
Analysis:
1) On the dive-site tab a filter is active.
2) Switching the tab to the dive-site-edit tab resets the filter.
3) This emits `filterFinished`
4) `filterFinished` is connected to `MapWidget::reload()`
5) `MapWidget::reload()` exits edit mode and therefore the flag can't be dragged.

I think 4) is wrong, but I'm quite certain that this was introduced to fix problems with the map-edit mode. This will probably take some thinking. That's all horrible spaghetti code!

bstoeger, 2019-04-29:
Actually I think 5) is wrong.

commit 23cf85e8 (2019-05-11): Map: don't reload map when in edit mode

During edit mode, we could get spurious reload() requests owing
to tabs being hidden. This led to undefined behavior:
In some cases entering dive site edit mode would show all dive
sites, in some only the dive site of the currently edited dive.

Therefore, refuse to reload the map while in edit mode. The
corresponding flag already exists.

Partially fixes #2076

Reported-by: Doug Junkins <junkins@foghead.com>
commit 743f2176 (2019-05-11): Map: don't call exitEditMode() in MapWidget::reload()

On reload of the map, the map exits edit mode. Sounds logical
at first, but the whole map-mode code is very unpredictable.

What happened was that when switching from the dive site table
to dive site edit mode, the code would enter map edit mode first.
Then, the dive site tab got its hide-signal, which would reset the
filter. This would reload the map and thus exit mode. Hence the user
can't drag the flag on the map.

Partially fixes #2076

Reported-by: Doug Junkins <junkins@foghead.com>
closed 2019-05-11 by dirkhh