subsurface/subsurface #2093: iOS App deleting unused dive sites

djunkins, 2019-05-15:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

In the master branch we now retain and save dive sites that have no dives associated with them, but in the current iOS release, any unused dive sites are deleted before saving to the cloud storage. This should be addressed when the next iOS release is out.

### Operating system:

iOS 12.2
iPhone 8

### Subsurface version:

Subsurface-mobile 2.1.6(4.8.4.66)

Official release

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-05-19:
I submitted a new build for review - once it has been approved you should get a notification via TestFlight. I'd appreciate feedback to make sure that none of the recent changes broke anything on iOS, once I'm confident about that I'll happily push this out as new version.

mikeller, 2026-10-17:
Hi @djunkins,

This is fixed: dive-site retention (keeping dive sites with no associated dives, rather than deleting them before saving to cloud storage) had already landed in master, and a new iOS TestFlight build was pushed shortly after your report specifically to bring iOS in line with that behaviour.

Closing this as resolved. Please reopen if a current iOS build still deletes unused dive sites on cloud save.

closed 2026-10-17 by mikeller