subsurface/subsurface #2096: Error message "Unable to Open Com3"

RedrumDiver, 2019-05-17:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Unable after Upgrade to Version 4.8.6 I can no longer update my dives from my oceanic Proplus 3 via USB. Dive computer connects to my PC and the Proplus 3 goes into PC connected mode. When I go do download to Subsurface it alerts to "Unable to Open Com3". I have reset the remembered computers and tired all my USB ports. 

### Operating system:

Windows 10

PC, Subsurface Version 4.8.6, Oceanic ProPlus3

### Subsurface version:

4.8.6

Worked before Upgrade to current version.

Official release and Updates

### Steps to reproduce:

Connect ProPlus3
Download Dives
Error message "Unable to Open Com3"

### Current behavior:

Cannot download dives
### Expected behavior:

Download dive logs
### Additional information:

### Mentions:

[image]

dirkhh, 2019-05-18:
Can you try the latest test version (it has quite a few changes in the way we deal with dive edits - nothing that should eat your dive log, but please be aware that this is not a small incremental change like the last few versions but a current build on the way to the next major version)
https://subsurface-divelog.org/downloads/test/subsurface-4.8.4-570-gd83b3e50c8fd.exe

bneuen, 2019-10-13:
This issue is still present in 2.9.3 (running Windows 7) when attempting to download/import from an Oceanic OCi.  This dive computer and cable work if backdated to a previous version of Subsurface (2.9.1).  Log file attached.

[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/3722032/subsurface.log)

jb2cool, 2020-03-08:
@bneuen there have some improvements in the underlying libdivecomputer since you posted this and there is a recent testbuild here https://github.com/Subsurface-divelog/subsurface/releases/tag/ci-release. Would you be willing to give this a test and let us know if this resolves your issue.

mikeller, 2026-10-17:
Hi @RedrumDiver,

This 2019 report doesn't have enough detail (which dive computer, which port, Windows version) for us to act on, and we've had no follow-up since. We're closing this as stale.

Please reopen with those specifics and a verbose log if you're still seeing "Unable to Open Com3" on a current build.

closed 2026-10-17 by mikeller