subsurface/subsurface #2101: Download from Shearwater Perdix not working with version 4.8.6 on Win10

Marianneee, 2019-05-19:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Perdix in connection mode.
Import from dive computer.
Filling in model and choosing bluetooth download mode.
"The local bluetooth adapter cannot be accessed"

### Operating system:

Windows 10, Swedish.

Shearwater Perdix Ai

The same happened on other computer running Win 7.

### Subsurface version:

4.8.6

I have only tried this version since the dive computer is brand new.
Downloading dives works perfectly in the Shearwater Desktop/mobile programs but ithey are not very good so I would like to be able to use Subsurface.

I am using official release.

### Steps to reproduce:

1) Starting Subsurface
2) Starting Perdix and starting bluetooth
3) Klicking import from dive computer
4) Choosing Vendor=Shearwater
Dive computer=Perdix Ai
5) Klicking "Choose Bluetooth download mode"
6) A window opens with "Remote Bluetooth device selection"
It is empty and I cannot click on "Scan" and at the bottom it says "The local Bluetooth adapter cannot be accessed"

### Current behavior:

No dives are downloaded.
### Expected behavior:

Downloading new dives into Subsurface.
### Additional information:

### Mentions:

dirkhh, 2019-05-19:
BLE under Windows is painful. Have you read the manual section about unpairing and repairing your Perdix Ai? It often takes multiple attempts.
Alternatively, consider downloading on Android or iOS (usually much more reliable) using Subsurface-mobile and syncing your dive list via the Subsurface cloud storage.

Marianneee, 2019-05-19:
Yes I tried unpairing several times. I have also downloaded the Android version, but it does not seem to work either.

dirkhh, 2019-05-19:
Since several of the developers have a Perdix Ai and use it successfully (especially, the developer of the Windows BLE support uses a Perdix Ai) I wonder what's different for you.
One thing we know is that the Perdix Ai has no concept of "unpairing". So if it thinks it's connected to something, it will not connect to a new device. Is it possible that it thinks it's connected to the Shearwater app on the Android device?
I know that this process is frustrating (been there, done it way too many times), but I can assure you that this is a well tested devices with Subsurface and that it should work fine on any of our platforms, once it is successfully paired...

k8smate, 2019-06-03:
In another thread it was acknowledged the released version has a bug that affects Windows ability to see any dive computer. An alternate version was suggested (with link) which lets Subsurface see my Perdix, but it cannot  download dives. I have tried unpairing and pairing, uninstalling all BT drivers then reinstalling, repairing with the Perdix but still the same behavior even though Subsurface can 'see' my DC, listed with the correct MAC address. Something different in what the developer is using and the actual released code I believe. As a retired IT pro I have seen this type of issue many times, sometimes due to development environment variables or ownership or links to build version source code. Still a bug that needs to be fixed. 

I AM using the android version and syncing via the cloud but that is not the 'workaround' I prefer to have to use.

neolit123, 2019-06-04:
responded here:
https://github.com/Subsurface-divelog/subsurface/issues/2117#issuecomment-498884860

hhassey, 2019-11-06:
Can confirm same behavior with Debian Buster, and Subsurface 4.9.3

mikeller, 2026-10-17:
Hi @Marianneee,

Same situation as the related report (#2117) from the same 4.8.6/Win10 era: Bluetooth and download handling on Windows has changed substantially since 2019, and we have no further detail or recent reports of this. We're closing this as stale.

Please reopen with your current version and a verbose log if your Perdix still won't download on a current build.

closed 2026-10-17 by mikeller