subsurface/subsurface #2103: Download issue

Meg671, 2019-05-21:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I recently downloaded subsurface to use for a project. There was no problem downloading the program but when I open the program, it is cut off on a four sides. I do not have access to the options on the top and i need to be able to access them. 
### Operating system:
PC. Windows 10 64 bit. English

Dell XPS 

### Subsurface version:

Version 4.8.6
### Steps to reproduce:

1. Download the installation file.
2. Install using the wizard
3. Open application
4. Sides are cut off (picture below)
### Current behavior:

### Expected behavior:

### Additional information:

[image]

### Mentions:

jb2cool, 2020-03-08:
@Meg671 i'm currently looking through some old bug reports and this caught my eye, firstly because it's a bit of a weird issue but also because it's using a fairly old version of Subsurface, have you managed to try some of the newer versions of Subsurface (we are up to 4.9.3 now) and whether this problem has gone away for you now?

mikeller, 2026-10-17:
Hi @mygim,

This 2019 report doesn't have enough detail for us to act on (no screen resolution, Windows version, or screenshot of the cut-off window), and there's been no follow-up since. We're closing this as stale.

Please reopen with a screenshot, your screen resolution, and Windows version if you're still seeing the download window cut off on a current build.

closed 2026-10-17 by mikeller