subsurface/subsurface #2105: issue fedora 30

ealcaniz, 2019-05-24:
### Describe the issue:

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Issue when open Subsurface in Fedora 30
./Subsurface-4.8.6-x86_64.AppImage: error while loading shared libraries: libcrypt.so.1: cannot open shared object file: No such file or directory

### Operating system:

Fedora 30

### Subsurface version:

Subsurface-4.8.6-x86_64

Yes

official release

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

cyberdiamond, 2019-06-24:
I had the same issue and fixed it with:

sudo dnf install libxcrypt-compat

buxtonpaul, 2019-09-22:
I think this is likely the same as #2285  and will be fixed with pull request #2296

glance-, 2019-11-10:
We're packing our own openssl in the appimage nowadays so this is solved.

closed 2019-11-10 by glance-