subsurface/subsurface #2108: Can't download dives from DC to mobile with FTDI

satoschu, 2019-05-27:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

 I tried to download dives from Oceanic VT 4.1 to Samsung Galaxy A7 2018 and got "Out of memory" failures.

### Operating system:

Android 9

### Subsurface version:

4.8.4.577

### Steps to reproduce:

1.Choose download section
2.Connect DC
3.Choose DC Modell
4.Hit download 
5.Nothing happens

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

[image]

[image]

glance-, 2019-05-28:
We're not doing a perticaly good job at logging what got the Out of memory. It might be libdivecomputer, libftdi or libusb. Only a logcat will tell.

Can you set up adb over wifi and run adb logcat while you're testing? That will give us a better clue.

satoschu, 2019-06-05:
Hello glance-
Can you Support mehr a little please. How can I Setup an adb logcat over Wifi?

Best regards

glance-, 2019-06-09:
Download and install android sdk tools on another device.

Enable adb over network on your device, adb connect <ip>:5555 (or whatever port it will end up on), adb logcat , and redo the whole ftdi thing on your device while saving the output from logcat, and attach it here. Its usually a good idea to walk the log file and remove anything to private who might end up there.

https://developer.android.com/studio/command-line/adb contains more deatils.

soki90, 2019-06-10:
Hi there. I have the same issue, as I wrote in the forum recently also. I did  logcat , but  filtered  adb logcat | findstr subsurface . I hope I didn 't miss any relevant output. Let me know if you'd like different output. File attached. 
I believe this is the info that can help:
06-10 18:01:32.323 14700 15026 D /android/subsurface/core/serial_ftdi.c: INFO: failed ftdi_new()
06-10 18:01:32.323 14700 15026 D /android/subsurface/core/serial_ftdi.c: INFO: serial_ftdi_open() failed
06-10 18:01:32.323 14700 14747 D /android/subsurface/qt-models/messagehandlermodel.cpp: INFO: "37.320: Out of memory"
[subsurface.txt](https://github.com/Subsurface-divelog/subsurface/files/3272648/subsurface.txt)

Also there is a strange behavior with the combobox where the FTDI can be selected. If you toggle Off then On the Bluetooth when the computer is connected on ftdi, the ftdi option disappears from the list. Not sure if it is related, rather not, it is rather another bug.
Thank you.
Sorin

satoschu, 2019-06-17:
Hello, shall I also do a logcat with my device or is the one above enough for debugging?

Best regards

glance-, 2019-06-18:
I really can't figure out why ftdi_new() would fail. ftdi_new() tries to allocate memory for struct ftdi, which  is 112 bytes.

Samsung Galaxy A7 should have 4Gb RAM.

My guess is that there are more clues in that logfile, which might got filtered out. Could you post a manually filtered copy, or email me it? Just read it and look for obvious email/password things wich random apps tends to log in plain text.

adb shell dumpsys meminfo org.subsurfacedivelog.mobile should be interesting to look at to, if we're leaking memory or something.

soki90, 2019-06-19:
Hi, Anton.
I attached the logcat and  memory dump. I hope it helps.By the way, I registered to the beta versions on google play, so now the logs are from a beta version. 
Samsung 9+, Android 9, App 2.1.6 (4.8.4.577), Suunto D4i. The behavior is the same, not working. 
Thank you,Sorin

soki90, 2019-06-19:
Hi, Glance.
logcat and meminfo file sent on email. I hope they will help.

Thank you,
Sorin

glance-, 2019-06-19:
@soki90 I read the logfile from you, and its clear that its the security system on android that prevents libftdi / libusb from initing. We're using some unsupported accesses directly to /sys , which are blocked on some devices. This is clear from the audit log lines right after we try to init libftdi.

We have had plans on how to rewrite the code to fix this, but no one have gotten around to do it.

The whole "Out of memory" is just a red herring.

soki90, 2019-06-19:
@glance- thanx. I was guessing so , since it started right after updating to Android 9. Previously worked on Samsung 8 and 9+ with no problem. What I have noticed is that some  USB terminal apps I used successfully to test Suunto connection to Android 9 where showing an additional(or just different...)  dialog on connection, asking for specific access  to the Suunto sports something. Which I didn't get from Subsurface, it only asked me what app I want to use as default for the device connected. But the dialog was different that the usb terminal app. . 

@satoschu , sounds like the same issue, but probably your logs can show  some different things? 

Thanx.
Sorin

satoschu, 2020-01-20:
Sorry, I was out of the case for a long time. Will here be worked on and I can support?
Is FTDI working on anybody's mobile with Android 9?

dirkhh, 2020-01-20:
I do not expect this to work on any Android 8 or newer devices, and I do not expect that to change.

If this is an important feature for you. I suggest using a different dive log app.

/D

satoschu, 2020-01-24:
Hello Dirk, can you tell me why? Glance wrote it would be possible to get it work.

dirkhh, 2020-01-25:
Correct. And the moment he actually implements this, I'll change my mind.
Please recognize, this is not a complaint about Anton - he has done amazing work on Subsurface over the years.
But this task has been on our radar for something like three years. I have spent way too many hours trying to figure out how to do this. I don't see anyone actively working on it, so to me as the maintainer the more honest path appears to be to tell people not to expect that this will ever happen. Because at this point that's my expectation.

The future are BLE dive computers. Almost all new dive computers coming out support BLE. FTDI is the past.

glance-, 2020-01-25:
I've bin busy with other things, and no I don't have a time table on when I'd get this done. Maybee in a couple of weeks, maybe never.

closed 2020-03-07 by dirkhh

soki90, 2020-03-08:
Hello. Version 2.3.0(4.9.3.1032) works now with my Suunto D4i on usb- serial connection on Samsung S9+ with Android 10. Thanks for your support!

PR #2650 (2020-03-07): Android serial clean
This pull request allows to use a virtual "usb-serial-for-android" device to download dives from divecomputers with different USB-to-Serial chipsets.

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [x] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
This pull request allows to use a virtual "usb-serial-for-android" device to download dives from divecomputers with different USB-to-Serial chipsets. With this changes, FTDI-Based divecomputers also work a [... cut]