subsurface/subsurface #2117: Win10 Bluetooth "local bluetooth adapter cannot be accessed" since 4.8.6

dm9876, 2019-06-01:
### Describe the issue:

- [x] Bug

### Issue long description:

Exactly the same symptom as issue https://github.com/Subsurface-divelog/subsurface/issues/1831

Except that it was introduced by 4.8.6 if I install 4.8.5 then bluetooth is accessible again.

[image]

Bluetooth in windows is enabled, the device is paired. I have tried disable / enable . unpair / re pair etc. All the while, Shearwater desktop can access the dive computer vs bluetooth without trouble.

### Operating system:

- Windows 10 Version 1809 (OS Build 17763.529)
- Shearwater Petrel 1 (v65)

### Subsurface version:

Running official releases 4.8.6, issue does not occur on 4.8.5 can reliably get it to work / not work by install each version.

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

I can work around this in windows by linking a Com port as outgoing bluetooth connection to the petrel. Then using the com port in Subsurface (ie not expecting subsurface to do any thing with bluetooth)

[image]

### Mentions:

 @neolit123  I see you fixed issue https://github.com/Subsurface-divelog/subsurface/issues/1831 so maybe you have some idea?

neolit123, 2019-06-01:
> I can work around this in windows by linking a Com port as outgoing bluetooth connection to the petrel. Then using the com port in Subsurface (ie not expecting subsurface to do any thing with bluetooth)

interesting workaround.

> @neolit123 I see you fixed issue #1831 so maybe you have some idea?

i did not fix the issue. it is still open.

unfortunately i've updated Windows 10 last night, and now my Bluetooth driver is not working.
i need to figure out how to make it work again.

neolit123, 2019-06-01:
i was able to reproduce the BT problem with 4.8.6.

you can temporary use this installer from our continuous build:
https://github.com/Subsurface-divelog/subsurface/releases/download/continuous/subsurface-4.8.4-591-gba31c56c7854.exe

it says .4, but infact it's the latest version.

@dirkhh seems that if i replace `Qt5Bluetooth.dll` in the 4.8.6 installer, with the one i build from `wip/win` it works too. possibly there is a difference between the CI build and the release build.

k8smate, 2019-06-03:
I too have been experiencing this problem and tried to install the build recommended in the last comment
(https://github.com/Subsurface-divelog/subsurface/releases/download/continuous/subsurface-4.8.4-591-gba31c56c7854.exe)
So now my Perdix ai is SEEN by the program (complete with address) but when I try to download I get either failed to download or cannot find my dive computer. I am rather new to this program and like the layout. Just FYI - I am having the same issue with the Shearwater PC program. It connects to my Perdix ai but fails when trying to download the list of dives. 
This all worked fine before the new Windows 10 update. 

I DO have my tablet working with Subsurface, so I can download dives and sync via the cloud, but it is time consuming since I really don't want to enter a lot of data via my tablet screen.

neolit123, 2019-06-03:
i had to update my BT driver after updating Windows 10.
also try removing / pairing the device just before Subsurface download.

neolit123, 2019-06-04:
the TL;DR for BTLE on Windows is that it has major stability issues.

for example my current Windows version is at:
```
>ver
Microsoft Windows [Version 10.0.17134.765]
```
i was able to download from the Perdix *only* after updating to the latest version of Windows 10 some time ago and finding the latest driver for the BT chip.

it now consistently works if the device is first removed and then paired before each download from Subsurface.

unfortunately the state of BTLE on Windows as an OS is not great, and there isn't much the Subsurface team can do for all combinations of PC / DC chipsets out there.

flashadams, 2019-07-04:
Saw identical behavior in win10 (1809). in 4.8.6 it couldn't interact with bluetooth adapter.  Reverting back to 4.8.5 allowed it to scan/interact with the perdix, however dives would fail to download.

dm9876, 2019-07-04:
For what its worth, the workaround I mention in the bug report seems to be more reliable for me than it was even in 4.8.5 and earlier. Maybe worth to get it mentioned in the doco? though I don't know if it is equally as workable with other dive computers (mine is Shearwater Petrel 1, which is bluetooth classic, not BLE)

mikeller, 2026-10-17:
Hi @dm9876,

This report is from Subsurface 4.8.6 in 2019, Windows Bluetooth handling has changed substantially across many releases since then. With no further detail and no corroborating reports in years, we're closing this as stale.

Please reopen with your current Subsurface/Windows version and a verbose log if you're still seeing the local Bluetooth adapter reported as inaccessible.

closed 2026-10-17 by mikeller