subsurface/subsurface #2120: Feature request: Allow Youtube/Vimeo links as dive media (using the API)

atdotde, 2019-06-03:
We have the possibility to add dive media via URLs. We could try to identify URLs pointing to Youtube/Vimeo and be smart about those, i.e. using their APIs, e.g. to extract Thumbnails. See https://developer.vimeo.com/api/upload/thumbnails

### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2026-10-18:
Hi @atdotde,

This is a reasonable nice-to-have, but a lower priority given most users attach photos directly rather than linking external videos, and it'd need integration with two separate third-party APIs (YouTube and Vimeo) plus thumbnail caching. Still wanted? If so we'll keep it on the list, just not an immediate priority.

notionparallax, 2026-11-10:
I put youtube links into my free text area. I wonder if it would be enough to just parse them into a clickable link?

mikeller, 2026-11-10:
[image]

notionparallax, 2026-11-11:
I had completely missed that the urls got a ctrl click edit tooltip. I was chatting to claude about the options, and it got very carried away and made a lot of different options

I wonder if it's worth adding the blue text with  `QSyntaxHighlighter` but leaving the rest of the behaviour as is.

My favourite is _Favicon link, URL while editing_ but it's way too much work and doesn't actually resolve the question that @atdotde posed.

I wonder if there's another option, where [oembed](https://oembed.com/) gets the tile image, and adds it to the media library, and then the tile becomes a link? 

Anyway, enjoy the fruits of my burning tokens:

| Tier | Qt building block | Extra editing UI | Network | How much work?|
|---|---|---|---|---|
| 1 Plain text | `QTextEdit` | None | None | |
| 2 Ctrl-click text | Event filter on viewport | Minimal | None | In Subsurface now |
| 3 Styled link | `QSyntaxHighlighter` | Minimal | None | Small change |
| 4 Favicon chip | `QTextObjectInterface` | Menu | Title + icon fetch | Moderate |
| 4b Chip, URL while editing | `QTextObjectInterface` + swap on cursor move | Minimal | Title + icon fetch | Moderate+ |
| 5 Preview card | Widget strip below the field | Buttons | oEmbed + thumbnail cache | Moderate |
| 6 Media embed | QtWebEngine | Player | Live player | Not really feasible, needs a web interface which adds 100mg to the package size |

### 1. Just text

https://github.com/user-attachments/assets/2375fba4-5bc2-4dc9-baaa-94280992396c

The URL is ordinary text. You copy it and paste it into a browser. It's edited like any other text.

### 2. Ctrl-clickable text (already in Subsurface)

https://github.com/user-attachments/assets/1072abcb-444f-49d6-b64a-4a2f4cf98b86

It looks like plain text. Hovering shows a "Ctrl+click to visit ..." tooltip, and Ctrl+click opens the browser (Cmd on Mac). An ordinary click edits. This is `TextHyperlinkEventFilter` in `desktop-widgets/simplewidgets.cpp`. It finds the URL under the mouse when you hover or click, so nothing extra is saved. The weak spot is that you only find out it's a link by hovering over it.

### 3. Styled link

https://github.com/user-attachments/assets/02c310fd-4eb3-46bf-b2a3-0bb934cb2c3a

Blue and underlined, like on the web. A plain click edits and Ctrl+click opens, the same convention as code editors, which settles the "how do I edit a blue link?" question. A `QSyntaxHighlighter` recolours URLs on screen only, so the saved notes don't change and the notes never switch into HTML mode. It's about one class plus a URL pattern, with no network.

### 4. Favicon link

https://github.com/user-attachments/assets/42db0f5b-4033-457c-bdee-92f601f6ac29

An inline chip showing the site's icon and the page title. The chip counts as one character: Backspace deletes it whole and the cursor skips over it. It needs a menu (Open link, Edit URL, Show as plain text, Remove link). `QTextObjectInterface` paints the chip in place of a single placeholder character, and when the notes are saved the chip is turned back into the URL.

The cost is fetching titles (oEmbed for video sites, the page `<title>` otherwise), plus a cache, an offline fallback, and probably a setting, because looking up titles contacts third-party sites whenever a dive is opened.

### 4b. Favicon link, URL while editing

https://github.com/user-attachments/assets/ff17c93f-a493-4ae8-9284-34e8bc435922

This is the same chip, but when the text cursor moves into it (by click or arrow keys) it turns back into the raw URL, which you edit like any other text. When the cursor leaves, it folds back into a chip. Obsidian and Typora call this "live preview". The URL itself is the editor, so no menu is needed. You still Ctrl+click to open.

Implementation: on `cursorPositionChanged`, swap the placeholder character back to the URL text, and swap it in again when the cursor leaves. The hard parts:

- The swaps are edits to the document, so they would go into the undo history unless kept out. That probably means managing undo ourselves.
- Cursor and selection positions shift by the URL's length minus one, which affects selections and search.
- The line reflows when the chip expands, so the trigger has to be the text cursor, never the mouse pointer. Otherwise the text would jump around under the mouse.

### 5. Preview card

[image]

Thumbnail, title and channel in a strip under the text box, with "Open in browser" and "Add to dive media" buttons. The text keeps the plain or blue link. It goes below the box rather than inside it because a `QTextEdit` can't contain real widgets. A card inside the text would have to be painted by hand, with buttons that only pretend to work. Below the box, it's ordinary Qt widgets. One oEmbed request gives the title, author and thumbnail URL, and the thumbnail can go in the same cache used for dive photos.

### 6. Media embed (not practical)

[image]

An embedded player is an `<iframe>`, which needs a full browser engine. That means QtWebEngine (Chromium), which the Qt6 Windows package already leaves out because of its size (100+ MB per DLL). QtMultimedia can play local files but not YouTube or Vimeo streams. The practical limit is a thumbnail with a play button that opens the browser, which is tier 5.

[... 5 lines cut ...]

## Which parts of oEmbed Qt can use

Earlier in this thread, the concern was that this would need two separate third-party APIs. YouTube and Vimeo both offer oEmbed endpoints that don't need an API key:

[... cut ...]
  "thumbnail_width": 480,
  "thumbnail_height": 360,
  "html": "<iframe width=\"200\" height=\"113\" src=\"https://www.youtube.com/embed/...\"></iframe>"
}
```

- **Usable:** `title`, `author_name`, `provider_name` and `thumbnail_url` are plain text and an image URL. Qt can show these in a chip (tier 4), a card (tier 5) or a dive media thumbnail.
- **Not usable in a `QTextEdit`:** `html` is the player itself. Qt's rich text understands formatting, tables, images and links, but not iframes or scripts.

mikeller, 2026-11-11:
AI-drafted; reviewed by Michael.

Thanks for the write-up and the mock-ups. They make the trade-offs easy to compare.

### My assessment

Two constraints matter a lot to me here:

- Many users dive in remote places or from boats with no internet, so the feature has to work well offline.
- I want desktop and mobile to behave the same way. Generating thumbnails on desktop only, and showing unintuitive links on mobile, isn't a good result.

By that measure, anything that needs a network lookup when a dive is displayed (tiers 4, 4b and 5, and an oEmbed thumbnail) is the weaker option. I'd rather start from something that is plain text, works offline and can behave the same on both platforms.

### AI-assisted technical analysis

What I found in the current source:

- **Desktop notes.** The only link handling is the Ctrl+click and tooltip in `TextHyperlinkEventFilter` (`desktop-widgets/simplewidgets.cpp`). It is desktop-only and nothing is stored, so a `QSyntaxHighlighter` (tier 3) would change how notes look but not the saved data.
- **Mobile notes.** Notes are shown in a `Controls.Label` with `Text.RichText` (`mobile-widgets/qml/DiveDetailsView.qml`) and there is no link-activation handling anywhere in the QML. From the code, a pasted URL looks like plain, inert text on mobile. I haven't run it to confirm.
- **Mobile media.** `mobile-widgets` has no picture or media code at all, so a "video as dive media" entry would currently have nowhere to show up on mobile.
- **Thumbnails and the file format.** `struct picture` stores only `filename`, `offset` and `location`, and `save_picture` in `core/save-xml.cpp` writes only those. Thumbnails are never in the dive file. `ImageDownloader` caches them per device (keyed by a hash of the URL), so a thumbnail fetched on one machine does not reach another. The release notes also say binary picture data was deliberately removed from cloud storage because it made syncs very slow.
- **Offline and oEmbed.** The oEmbed lookup fails without internet, and a successful lookup is only cached on that device. That is the mismatch with both constraints.

Options that fit those constraints better:

1. **Make links work the same everywhere, with no network.** Desktop: tier 3 blue links (click to edit, Ctrl+click to open). Mobile: make URLs in the notes tappable. Neither needs a file format change or any network access, and the two could be separate small PRs.
2. **Capture something at add-time, if a richer entry is wanted.** For example, a user-typed title (or one fetched once, while the user happens to be online) stored as text next to the link. A small title costs almost nothing to store and would show identically on both platforms, with or without internet. Embedding thumbnail image data in the dive file runs into the cloud-sync concern above, so I'd avoid it. Any new attribute needs checking for how older Subsurface versions handle it before it goes into the file format. I haven't verified that.

Option 1 looks like a good first step on its own. Option 2 would need a design discussion here first, mainly about where the title lives and how it round-trips through the file format.

If you want to try any of this, please open a PR against `subsurface/subsurface` and keep each piece a separate change. I can't promise acceptance or a release timeline, but small, self-contained changes are the easiest to review.