subsurface/subsurface #2138: Merging dives discards data

mturkia, 2019-06-26:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

At least duration, cylinder and watertemp are discarded when merging dives manually.

### Operating system:

Linux, (Ubuntu 19.04).

### Subsurface version:

git master (checked out and compiled about a month a go)

### Steps to reproduce:

Download new data from DC and trigger manual merge. I.e. download to new trip, remove the dives from a trip, and then manually merge to a dive from different DC.

### Current behavior:

Some snippets from my git diff shows unwanted changes on my dive log.

First one looses divesiteid
`
--- a/2019/01/14-Fiji/20-Sun-08=08=49/Dive-1108
+++ b/2019/01/14-Fiji/20-Sun-08=08=49/Dive-1108
@@ -1,5 +1,5 @@
 duration 76:45 min
-divesiteid 888f6ffc
 divemaster "X"
 buddy "X"
 cylinder vol=11.094l workpressure=206.843bar description="AL80" o2=31.0% depth=41.47m
+watertemp 28.0°C
`

In this one, dive duration is moved to divecomputer specific file and cylinder information is lost from the Dive.
`
--- a/2019/01/14-Fiji/20-Sun-08=08=55/Divecomputer
+++ b/2019/01/14-Fiji/20-Sun-08=08=49/Divecomputer-003
@@ -3,6 +3,7 @@ deviceid 20993833
 diveid 1fd6de1f
 date 2019-01-20
 time 09:08:55
+duration 76:40min
 maxdepth 22.22m
 meandepth 14.181m
 airtemp 30.0°C
diff --git a/2019/01/14-Fiji/20-Sun-08=08=55/Dive b/2019/01/14-Fiji/20-Sun-08=08=55/Dive
deleted file mode 100644
index 0b77e15..0000000
--- a/2019/01/14-Fiji/20-Sun-08=08=55/Dive
+++ /dev/null
@@ -1,2 +0,0 @@
-duration 76:40 min
-cylinder vol=11.094l workpressure=206.843bar description="AL80" o2=31.0% depth=41.48m
`

### Expected behavior:

No existing data is lost when merging new DC to a dive.

### Additional information:

### Mentions:

bstoeger, 2019-06-30:
Could you perhaps export the two dives in question (pre-merge)? There is an anonymization function if you don't want to make any private data public. I tried it with some test data and I didn't see any obviously strange behavior.

Thanks.

mturkia, 2019-07-01:
I have failed to create an example log that would replicate the issue. The problem actually occurs on older dives than the ones that are being merged. The bug hit me at least twice during the last trip, but now I cannot re-create it. It also could be related to something else than merging the dives, but that seemed quite plausible explanation at the time I noticed the issue.

mikeller, 2026-10-17:
Hi @mturkia,

Thanks for reporting that merging dives discards data. We haven't been able to reproduce this against current merge logic, so we'd need a specific sample, ideally the two dives (exported files) you merged, showing what data existed beforehand and what went missing afterward.

If we don't get a reproducible sample within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.