subsurface/subsurface #2143: Bug: User manual serach functionality

willemferguson, 2019-06-27:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Within the webkit browser on can search for a term in the manual by hitting Ctrl-F. This brings up a search box which should look as follows:
[image]

There are three buttons (^=previous occurrence of search term, v=next occurrence of search term, x=close search)

With latest master, when hitting Ctrl-F, I get:
[image]

The up and down arrows are missing. This severely limits the scope of the search functionality in the user manual.

### Operating system:
 Ubuntu 18.04

### Subsurface version:
 4.8.4-659

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-07-04:
Hmm, no idea what caused that. This works when I test it on my Mac
Need to try it in my Ubuntu VM when I'm back at home

mikeller, 2026-10-17:
Hi @willemferguson,

Thanks for flagging a problem with the user manual's search functionality. Could you give us specifics, what you searched for, what you expected to find, and what actually happened (no results, wrong results, a broken link, etc.)? That'll help us tell whether this is a content issue or a genuine search-feature bug.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.