subsurface/subsurface #2144: Bug: Profile when adding dive by hand in Win (again)

willemferguson, 2019-06-27:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

This bug was reported during the last 2 weeks and was ostensibly fixed. However, in Windows it has not been fixed. At startup a credible profile is shown after one selects Log -> Add dive (15 m for 40 min). If the duration of the dive is changed in the Notes Tab (e.g. 40 -> 55 min) the profile becomes crazy with extremely low rates of descent and ascent. 
[image]

Even when forcing the profile to be realistic by hand-editing the profile itself using the white dots, the slow ascent from the last reference point (usually 5m) to the surface remains.

### Operating system:
 Win 7

 Not on Linux

### Subsurface version:
 V 4.8.6-659.
 Yes in Win but not in Linux
 Continuous build

### Steps to reproduce:

Open Subsurface
Log-> Add Dive
Change dive duration from 40 min to 55 min.

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-07-04:
There must be some uninitialized variable problem going on here.
I tried it in a Windows VM and it worked for me, but typically when things like this randomly fail it's uninitialized variables that cause us pain... I'll try some more

jb2cool, 2020-03-08:
Interestingly i can actually reproduce this error on the latest stable release, on Linux it works just fine but on Windows the dive profile gets redrawn with a 1m/min descent and ascent.

dirkhh, 2020-03-08:
Can you build Windows binaries (fairly easy with the container setup). Then you could instrument this code and add some logging information...

jb2cool, 2020-03-09:
How frustrating, I've tested this on one of the laptops at work and like @dirkhh has experienced it's working just fine there. These are both naively running Windows 10 and both on the latest stable release of Subsurface (4.9.3). I might just try a simple uninstall/reinstall to see if that straightens it out.

jb2cool, 2020-03-09:
Found it and it's reproducible on Linux as well.

When you add the dive the descent and ascent rates that have been set in the planner are respected but when you edit the dive parameters (either depth or duration) then only the 6m to surface rate is used to draw the profile of both the descent and the ascent.

The reason i hadn't seen this previously is that my Windows machine is my main machine so that's where i do my planning and the other devices (Linux laptop and Windows laptop at work) are more read-only.

jb2cool, 2020-03-09:
Notes for my learning rather than real conversation

The first drawn dive (15m for 40min) gets generated by desktop-widgets/mainwindow.cpp.
I think (although not 100% sure) the redrawn dive (after you've edited it) gets created by core/device.c

I just don't understand where the redrawn dive is pulling the 6m to surface ascent rate. You can tell it IS pulling the 6m to surface ascent rate by adjusting this and then adding/editing a new dive.