subsurface/subsurface #2146: Bug: Subsurface won't record data accurately when entering by hand

sol0712, 2019-06-27:
I'm running Subsurface 4.8.6 on MacOS 10.14.5. I've been trying to enter the duration by typing 36:00. I tried entering just 36 as well, but no matter what I do, it just pops back to 12 minutes the instant I try to save, or hit enter, or tab over to the next field. It simply will not let me save a dive time of 36 minutes.

Screenshots are attached.

[image]
[image]
[image]

sol0712, 2019-06-27:
It's a pattern. enter value in Field 1. Save. Enter Value in Field 2. Save. Field 1 changes to something else all on its own ...

bstoeger, 2019-06-30:
That's weird. Release 4.9 is coming soon - could you perhaps check current master? This part of the code was rewritten quite extensively.

sol0712, 2019-07-01:
I'd be happy to, if I knew what that meant and what to look for. I'm not a
developer, sadly. Will keep an eye out for 4.9 in the meantime.

_____________________
linkd.in/1K9tIZT
solchrom.com

dirkhh, 2019-07-04:
Go to https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous and download the Asset named `Subsurface-4.8.4-xxx-gxxxxxx.app.zip` (the `xxx` will change every time we push changes into our master branch). Unzip the file and run `Subsurface-4.8.4-xxx-gxxxxxx.app/Contents/MacOS/Subsurface` to test a current test build. Replacing `xxx` with the correct name in the downloaded file, of course.

Sometimes right after a change was pushed that `.app.zip` will be missing. If that's the case, check back in a few minutes, it's probably building right now. If it's still missing after a few tries, contact me so I can check what's going wrong...

sol0712, 2019-07-05:
Hi Dirk:

Thanks for your message.

So I've tried that. The newest test version, if that's what it is, seems to
be working, but I can't say so for sure because I haven't had an
opportunity to add dive data manually. Or use the data cable (a story on
its own).

The real fun part? I usually dive with a Mares Icon HD. I can't recall
whether or not i managed to make it work with Subsurface because I use it
with Mares proprietary software DiversDiary. Usually works OK, but the
computer's off getting the battery replaced (which takes a ridiculously
long time), so in the interim, my dive shop's letting me use a Suunto Zoop
Novo. I had to install god knows how much crap on my MacBook just to get
Suunto's proprietary DM5 software working -- to no avail.  Eventually I got
the Zoop Novo and Subsurface talking to each other, but that was an ordeal
and a half. Just hooking the Zoop up to its data cable was an adventure.
Can't wait for the day my partner lets me buy a Shearwater.

Sol

______________________
linkd.in/1K9tIZT
solchrom.com

jb2cool, 2020-03-08:
@sol0712 i'm looking through some old bug reports currently to try and clear some that have been resolved now. Did you ever manage to confirm that this issue is now fixed for you?

sol0712, 2020-03-08:
Hi Jason, and thanks for your message. I don't remember all the details,
but given that I haven't emailed you and your team since last July, we can
probably take it that the matter was resolved well enough. I am still
having trouble entering tank sizes and pressures accurately, but the
variance is small enough that I can live with it.

Sol

______________________
linkd.in/1K9tIZT
solchrom.com

bstoeger, 2020-05-03:
Original bug seems to be fixed - closing.

> I am still
having trouble entering tank sizes and pressures accurately, but the
variance is small enough that I can live with it.

I couldn't reproduce that either with metric or "imperial" settings. If it still happens, please open a report with a test case.

closed 2020-05-03 by bstoeger