subsurface/subsurface #2149: Feature request: add export/import of locations

kittykat, 2019-06-30:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:
When I do a liveaboard with buddies who also use Subsurface, I would like to be able to share the dive site data with them so that we're not all having to enter it manually as that is quite tedious for 20+ dive sites.

### Operating system:
For the desktop version

### Subsurface version:
4.8.6

### Steps to reproduce:

### Current behavior:
It's not possible to export or import locations.

### Expected behavior:
I would like to be able to export all locations or specific ones that could then be imported by another user.

### Additional information:

### Mentions:

bstoeger, 2019-06-30:
If I remember correctly, @djunkins implemented that recently...? There should be an "Import dive sites" option in the main menu.

djunkins, 2019-06-30:
@kittykat Exporting and importing of dive sites are in the current master branch. I believe they will be in the 4.9 general release when it comes out.

kittykat, 2019-07-01:
@djunkins Can confirm it works from master! Really happy to see the feature

closed 2019-07-01 by kittykat