subsurface/subsurface #2152: Bug: minimum window size is too tall

kittykat, 2019-07-01:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The build from master won't maximise for me (works fine in 4.8.6) as it looks like the minimum window size is taller than my screen (using an XPS13). I can't make the window any shorter so I can't see the whole window on my screen.

### Operating system:

Linux desktop (GNOME + Wayland)

### Subsurface version:

4.8.4-663-g5c1446a87acb

### Steps to reproduce:

1) Start up subsurface
2) Try to maximise the window (using usual methods)
3) Try to make the window shorter by grabbing and dragging top edge

### Current behavior:
Window won't maximise and can't be made shorter.

### Expected behavior:
Minimum window size is smaller than my screen so that I can see all content at the same time and maximise the window.

### Additional information:

### Mentions:

dirkhh, 2019-07-04:
What is your screen resolution?
Playing around with things it does seem that there's a fairly massive minimum height. I wonder when then got introduced...

closed 2019-07-04 by dirkhh

kittykat, 2019-07-04:
> What is your screen resolution?

3200×1600 and I'm using 200% scaling

Thanks for the fix, I can see the whole window now.

Slightly different issue with the filter section though (it doesn't change size depending on how much space I give it):
[image]

PR #2157 (2019-07-04): Desktop: reduce minimum vertical size
The Filter widget really needed to be scrollable as it is very tall.

Fixes #2152

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Changes made:

Simply add a scroll area around the widget
commit 774e9ae1 (2019-07-04): Desktop: reduce minimum vertical size

The Filter widget really needed to be scrollable as it is very tall.

Fixes #2152