subsurface/subsurface #2153: New feature: better bottom time reporting

kittykat, 2019-07-01:
### Describe the issue:
- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:
I want to be able to check my total bottom times for:
* only open circuit (definitely excluding freediving and preferably excluding CCR/PSCR - I have entries in all of those categories)
* all dives in one trip (multiple dive sites)
* all dives at one dive site
* dives until a specific date (to double check that my maths is correct in my paper logbooks which I still need to use for now)

The first of those three is the most important as that's what I need to know most often.

### Operating system:
Linux Desktop version

### Subsurface version:
4.8.6
4.8.4-663-g5c1446a87acb

### Steps to reproduce:

### Current behavior:
The only bottom times I can check at the moment are per month/year and total bottom time (and by depth/temperature from master).

### Expected behavior:
I would like to be able to select some dives and view the total times only for those. E.g. from a menu item "Selected dive statistics" next to "Yearly statistics"

### Additional information:

### Mentions:

atdotde, 2019-07-01:
Most of that is already there: When you select more than one dive in the dive list, the statistics tab shows that aggregate information you are looking for.

mikeller, 2026-10-18:
Hi @kittykat,

Following up on atdotde's point above: selecting multiple dives already shows aggregate statistics for that selection in the Statistics tab, which covers the trip/dive-site/date-range groupings you asked about via existing dive-list filtering and multi-select. Closing this as mostly addressed by that existing capability.

The one part not explicitly confirmed is cleanly excluding freediving/CCR/PSCR dives from an OC-only SAC-rate view, if that specific filter isn't available in the current dive-list filter UI and you still want it, please reopen with that specific ask and we'll look at it as a smaller, separate filter-criteria addition.

closed 2026-10-18 by mikeller