subsurface/subsurface #2154: New feature: track pool dives

kittykat, 2019-07-01:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

For the training that I am doing right now, I need to track my pool sessions as well as my open water dives. Obviously my pool sessions don't count towards my total dive numbers or bottom time.
In general, outside of my current training, I also like to keep track of when I practice which skills in the pool and with whom.

### Operating system:

Linux desktop

### Subsurface version:

4.8.6
4.8.4-663-g5c1446a87acb

### Steps to reproduce:

### Current behavior:

I've been using "freedive" to mark both my freediving dives and pool session as I don't care about tracking freediving stats

### Expected behavior:

Have another "dive mode" or some sort of "tag" that can be used to track pool sessions which doesn't contribute to total dive numbers or bottom times (or can be excluded from the stats as a setting in the settings menu or a tickbox the stats dialog)

### Additional information:

### Mentions:

atdotde, 2019-07-01:
Same answer as on the other topic: Use the filter on tags and select the remaining dives. This should give you the statistics you are looking for. Alternatively, you could use a separate log book file for your pool dives.

bstoeger, 2019-07-03:
A dive mode does not make sense to me: A pool dive can be either OC, rebreather or freedive, no?

Perhaps really use the free-form tags + filter?

kittykat, 2019-07-03:
> A dive mode does not make sense to me: A pool dive can be either OC, rebreather or freedive, no?

True

> Perhaps really use the free-form tags + filter?

There's really two problems that I see:
1) If I log my pool dives, they count towards my total and therefore the numbering of open water dives does not match the ones I have to keep in paper form. e.g. if I was just starting training, my paper log open dive 1 would be something like dive 4 in subsurface if I'd had 3 pool sessions.
2) If I use tags and filters, I lose the "yearly stats" functionality (including all the shiny new features in the upcoming version) and have to rely on the "Statistics" tab

I'm not sure what the correct solution would be… ideally, I'd have a tickbox when editing a dive for "don't number or count or include this dive in any way other than keeping a record of it".

With the way I'm tracking pool dives right now (bit less than ideal), it would also work to have a "dive type" checkbox on the yearly stats page to filter there by dive types.

mturkia, 2019-07-03:
Under the preferences, there is the option to "display invalid" dives. IIRC this was created just for the purpose of having i.e. pool dives in the log, but not including them in the normal dive list. And a quick search in mailing list archive indicates that this is how the tagging of dives was actually started. Unfortunately I cannot remember how a dive is marked as invalid - or even if that is still possible. Maybe this would suffice? (If it works or we fix it.)

kittykat, 2019-07-03:
I can see the preference, but can't see any way to set it. What makes a dive "invalid" at the moment?

bstoeger, 2019-07-03:
> If I log my pool dives, they count towards my total and therefore the numbering of open water dives does not match the ones I have to keep in paper form. e.g. if I was just starting training, my paper log open dive 1 would be something like dive 4 in subsurface if I'd had 3 pool sessions.

You can renumber the dives.

> If I use tags and filters, I lose the "yearly stats" functionality (including all the shiny new features in the upcoming version) and have to rely on the "Statistics" tab

Enabling filtering for yearly stats shouldn't be hard.

> I can see the preference, but can't see any way to set it. What makes a dive "invalid" at the moment?

The feature is not currently implemented.

kittykat, 2019-09-02:
> You can renumber the dives.

I've found renumbering dives a bit of a pain as I have to scroll through past dives to figure out the last one. It didn't used to work reliably for me, but it seems to be ok with my current build.

> Enabling filtering for yearly stats shouldn't be hard.
>
>
> The feature is not currently implemented.

That sounds like it would be a good solution

bstoeger, 2019-12-13:
@kittykat: Please let us know if #2440 works for you (apart from the UI) and whether you need some adjustments.

bstoeger, 2020-05-03:
Mark dives as invalid is in current release. Please let us know if this fulfills your needs.

bstoeger, 2020-10-03:
No response in half a year - let's assume that did it and close.

closed 2020-10-03 by bstoeger