subsurface/subsurface #2162: Featrue request: allow deleting single connections to dive computers

hhmeyer, 2019-07-06:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x ] New feature request
- [ ] Discussion request

### Issue long description:
I added a Shearwater Perdix AI to my equipment and had to experiment with the Bluetooth connection (trying internal Bluetooth and the dongle Shearwater sells with the Perdix AI, than switching back to my OSTC) and now I have multiple entries for each computer. When I want to read out dive data I never know which button to click since the connections a not distinguishable by name.

[image]

### Operating system:

OSTC sport, Shearwater Perdix AI

### Subsurface version:

4.8.6 official

### Steps to reproduce:

1) open download dialog
2) add one or more new connections
3) get confused which connection worked and which did not work since the connections to the same computer a not distinguishable by name
4) find no way to delete a single connection
5) delete all connections using "Preferences" -> "General" -> "Clear settings" -> "Reset remembered dive computers"
6) start over

### Current behavior:

The only option is to delete all devices.

### Expected behavior:

Being able to select one or more devices and delete them while leaving other untouched, preferrably directly in the dialog.

dirkhh, 2019-07-06:
Good idea. We're working on wrapping up 4.9 right now and as this would change texts (and we are in the final stages of translation) this won't make it into that release, but it should be fairly easy to add for a release after that.

hhmeyer, 2019-07-09:
eliminated typo

mikeller, 2026-10-17:
This has been implemented: PR #4160 "Add delete dive computer buttons" added per-entry delete (X) buttons next to each remembered dive computer connection in the import window, so you no longer need to use the blanket "Reset remembered dive computers" in Preferences to remove just one.

Closing this as implemented. Please reopen (or file a fresh issue) if the delete buttons aren't present or don't work as expected on a current build.

closed 2026-10-17 by mikeller