subsurface/subsurface #2163: Force download of all dives does not work a expected

hhmeyer, 2019-07-06:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
After a dive I downloaded the dive (the latest, newest one) into the desktop version of Subsurface. Then I stupidly deleted the freshly imported dive before it had been saved in Subsurface.

As far as I know the only way to redownload that dive from the OSTC ist to force the download of all dives and then only select the wanted one(s) to download.

Fetching all dives took a while, at that time there were about 170 dives in the computer. The list with all dives appeared in the download dialog, all dives were selected for download.

I deselected all, but the UI did not change. Only after I scrolled down in the list the status was updated. Same goes for the other button "select all".

Ok, lets try this again. Download all dives, deselect all, select only the last one, click OK. Nothing happens. I look at the display of the OSTC, Bluetooth mode is no longer active.

After the download of the complete list of dives from the OSTC either something in the OSTC times out and it reboots or Subsurface tells the OSTC to reboot.

### Operating system:
Mac with macOS 10.13.6, 
OSTC sport with firmware 10.53

### Subsurface version:
Subsurface 4.8.6 official

### Steps to reproduce:
see long description above

### Current behavior:
UI does not refresh to represent user action.
Dive computer reboots and thus ends Bluetooth mode.

### Expected behavior:
The UI should refresh without being triggered to redraw by user interaction.
The dive computer should stay in download mode after downloading the list of dives.

### Workaround
The only way I found to redownload this dive was to use a backup of my Subsurface database from a few days earlier, download the dive into this database, export it, switch to my current database and import the XML file.

dirkhh, 2019-07-06:
This seems wrong... if you indeed lost the LAST dive, then the easy solution is to simply download again (without forcing a full download) as that should see that last dive as "new".
And the solution to having lost, say, the third from last dive is similar. Export the newest dive BEFORE the lost dive. Open Subsurface with that exported dive as your dive log. Download from the dive computer. Only dives newer than that dive will be downloaded.

Of course, all of these are *reasonable* (I hope) workarounds to more comfortably solve the problem of re-downloading. The actual bug that you report is something that we need to look into as well. Neither the UI issue nor the Bluetooth issue are OK. Bluetooth is unfortunately rather fragile and we run into a fair number of cases where problems are near impossibly to reproduce. But we definitely should be able to fix the UI issue.

Thanks for the report!

mikeller, 2026-10-19:
Revisiting this old report: thank you for filing it originally, and sorry it's taken this long to follow up.

We looked into the checkbox-refresh problem described here (the download dialog's dive list not visually updating checkbox state after deselecting dives, or after Select all/Deselect all, until the list is scrolled). On direct testing against the current codebase, the checkboxes update immediately, both on individual clicks and on Select all/Deselect all, so we were not able to reproduce the stale-checkbox symptom as described.

It's possible this was fixed incidentally by other changes to the dialog or to Qt itself over the years since this was reported (2019, Subsurface 4.8.6, macOS 10.13.6). It's also possible it's still present under specific conditions we haven't hit (for example a very large dive list, a particular platform, or a particular Qt version).

If anyone reading this is still seeing the checkboxes fail to refresh until scrolling, in a current build, could you let us know:

1. Exact Subsurface version (Help -> About).
2. Operating system.
3. Roughly how many dives were in the list when you saw it.
4. The exact steps: did it happen on individual dive clicks, on Select all/Deselect all, or both?

Separately, the Bluetooth dropout after a full-list download described in this report is a distinct, known Bluetooth reliability issue and is not addressed by this investigation.

Without a current reproduction we don't plan to carry a speculative fix forward for the UI symptom, but we're happy to pick this back up if someone can confirm it's still happening.