subsurface/subsurface #2167: mobile: SIGSEGV on BT download from DC on Android

janmulder, 2019-07-08:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Using the current master, and a self build (arm) Android APK, I have a reproducible SIGSEGV in `DiveImportedModel::repopulate`. Interestingly, I cannot reproduce this crash using mobile-on-desktop (which complicates debugging).

To reproduce: download from a BT/BLE DC. And look at the logcat, and see it crash just after the finish of a successful download thread.

### Operating system:
OS: Android 9, LineageOS 16.0 

### Subsurface version:
What version of Subsurface are you running? current master (4.8.4.681)
Does the same happen on another Subsurface version? not reproducible on 4.8.4.577
Are you using official release, test build, or compiled yourself? self compiled. 

### Steps to reproduce:
See above.

### Mentions: @bstoeger (could this be related to undo work?)

bstoeger, 2019-07-09:
Does it crash directly in `DiveImportedModel::repopulate()` or in one of the two function calls?

Because I don't see how this could crash in this function unless either the `this` pointer is invalid or there is some preceding memory corruption (which would explain why it doesn't happen on mobile-on-desktop).

Could you perhaps try running mobile-on-desktop with valgrind and see if there is some memory corruption reports? Thanks.

bstoeger, 2019-07-09:
Well, for me mobile-on-desktop crashes right at startup, so there seems to be some fundamental issue. :)

bstoeger, 2019-07-09:
Never mind - that was an unrelated issue (see #2168).

bstoeger, 2019-07-18:
@dirkhh: shouldn't we try to fix this before 4.9? Sounds like a show-stopper to me.

janmulder, 2019-07-18:
In the meantime, I did valgrind mobile-on-desktop, and saw nothing suspicious there (with this scenario). Notice, not the exact same SHA this issue is written against (but a little newer). Unfortunately, I do not have  access to my DCs (everything is seriously packed for a cave diving exploration starting very soon), so I cannot verify things right now. My understanding is that 4.9 desktop is already tagged, and build for release (so no more changes, unless ...). Further, 4.9 desktop is not per definition the release of 2.2.0 mobile.

Also, I did numerous BT/BLE downloads over the past couple of days, mainly using arm64 builds. And this issue is not there. I'm sure I found it on the arm build. So, I'm not sure it can be reproduced any more.

bstoeger, 2019-07-18:
> Further, 4.9 desktop is not per definition the release of 2.2.0 mobile.

Ok - I thought desktop and mobile follow the same release schedule. Removing release-blocker tag.

> Also, I did numerous BT/BLE downloads over the past couple of days, mainly using arm64 builds.

Sounds unlikely, but perhaps it is a 32 vs 64 bit problem?

dirkhh, 2019-07-18:
I have tried BLE downloads with both 32 and 64 bit with the code from #2182 and that seems to work.
Because I really wanted to get 4.9 tagged and built before heading on my next trip (which I am now on, writing this in Europe), I decided to merge that one after the release and do a Subsurface-mobile 2.2 beta afterwards to see if other problems pop up. I can try to do this while traveling maybe even later today (depending how tired I am)

closed 2020-12-13 by janmulder