subsurface/subsurface #2169: CSV Export: Water temperature column always empty

h3llra1ser, 2019-07-10:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When you export the dive details to CSV, the water temperature column always is an empty string

### Operating system:

Win7 + Linux, both behave the same

### Subsurface version:

V 4.8.6 official

### Steps to reproduce:

Export dive details (not the dive profile) as CSV and look at the water temperature column contents

### Current behavior:

empty output

### Expected behavior:

some temperature :)

### Additional information:

### Mentions:

bstoeger, 2019-07-10:
Doesn't happen with my log. But water temperature is strange, as an average of the dive computers is calculated. So it might have to do something with your particular log. Could you post a minimal example?

h3llra1ser, 2019-07-11:
Sure, picked a random dive and attached xml and csv.

[temperature_export_example.zip](https://github.com/Subsurface-divelog/subsurface/files/3379847/temperature_export_example.zip)

bstoeger, 2019-07-11:
Weird I have two logs from very different setups which work, but your's doesn't. And it's not immediately obvious where the difference is. Moreover, I don't understand how temperature is treated in `xslt/xml2csv.xslt`. Therefore assigning to @mturkia.

PR #2178 (2019-07-13): Fix special case for temperature in CSV export

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

The old code took air and water temperatures primarily from
dive/divetemperature or secondarily if the previous one didn't have
either one of them from divecomputer/temperature. The new code attempts
to take first air temperature from dive/divetemperature or
diveomputer/temperature and then on a separate run the water
temperat [... cut]
commit 96ee6898 (2019-07-13): Fix special case for temperature in CSV export

The old code took air and water temperatures primarily from
dive/divetemperature or secondarily if the previous one didn't have
either one of them from divecomputer/temperature. The new code attempts
to take first air temperature from dive/divetemperature or
diveomputer/temperature and then on a separate run the water
temperature from these both. Thus we should be fine if one temperature
is in dive/divetemperature and the other in divecomputer/temperature.

Fixes #2169
closed 2019-07-13 by dirkhh