subsurface/subsurface #2170: Export of dive profile images (PNG) problem when exporting multiple dives

h3llra1ser, 2019-07-10:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When you export profile images of multiple dives all profile images have the same Y axis and also wrong scale and ticks of the X axis.
It seems that the grid gets initialized only for the first exported dive (which equals to the profile shown in subsurface) - but the other profiles dont match the axis ticks.
However, the profile grap itself is drawn correct if you ignore X and Y axis in the image :)

### Operating system:

Win7 and Linux - both behave the same

### Subsurface version:

V 4.8.6 official

### Steps to reproduce:

### Current behavior:

X and Y axis correct for the first exported dive, all others have the same axis set which is wrong:

### Expected behavior:

The X and Y axis should be reinitialized for each PNG export in the same loop

### Additional information:

### Mentions:

atdotde, 2019-07-10:
I cannot confirm this with the current development version. You you please try it with a binary from https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous

atdotde, 2019-07-10:
Checking again, it seems for me, only the y axis is concerned.

h3llra1ser, 2019-07-10:
Testes again with
Subsurface-4.8.4-679-g9eb860d45d65-x86_64.AppImage

same here....
Yes, as i wrote, its only the axis labels and ticks which are wrong

atdotde, 2019-07-10:
This is weird. Somehow the profileYAxis is not recomputed. Could not figure out why. Have to run.

bstoeger, 2019-07-10:
Try setting the "animation speed" in the preferences to 0 (i.e. no animations). I have a feeling that the image is written before the animation has finished.

To me it seems fundamentally flawed that the profile widget of the main window is reused.

bstoeger, 2019-07-10:
I'm not sure that my analysis is correct. If it is, it is hopefully fixed by #2171, so please check that out.

h3llra1ser, 2019-07-10:
Tested again with a set of 50 dives.
Disabling animations (set speed to 0) fixes this issue!

bstoeger, 2019-07-10:
Thanks - hopefully #2171 fixes the issue even without disabling animations.

atdotde, 2019-07-11:
It does

commit efe9bcce (2019-07-12): Profile: stop animations on export

The plotDive() function had a flag to plot pictures asynchronously.
This was used on export. Rename this field to "instant" and disable
animations when set. This should make sure that the axes are properly
exported.

Fixes #2170
closed 2019-07-12 by atdotde