subsurface/subsurface #2179: App: Entering invalid duration cancels input for new dive

sjvudp, 2019-07-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When entering the duration for a new dive in a way the user expects, but not subsurface (i.e. "0:41" for 0 hours 41 minutes), input is canceled and all data entered is lost.

### Operating system:
Android 8.0
Huawei P10 Lite

### Subsurface version:
2.1.6
Happened in older versions, too

### Steps to reproduce:
1) Add a new dive, entering values as you like

2) Enter "0:41" as duration

### Current behavior:
Input is aborted without message

### Expected behavior:
Input should not abort, format should be accepted

### Additional information:

### Mentions:

dirkhh, 2019-07-14:
I can't reproduce this. 0:41 is accepted for me as "41 minutes"

sjvudp, 2019-07-15:
>>> Dirk Hohndel <notifications@github.com> schrieb am 14.07.2019 um 02:15 in
Nachricht <Subsurface-divelog/subsurface/issues/2179/511162799@github.com>:
> I can't reproduce this. 0:41 is accepted for me as "41 minutes"

Amazingly I couldn't reproduce as well, but it happened to me several times. Maybe the conditions to trigger are more complex. I'll retry to find out, hopefully.

mikeller, 2026-10-17:
Hi @sjvudp,

Thanks for reporting that entering an invalid duration cancels your input on a new dive. Could you give us the exact steps, which field, what value you typed, and what happens on screen? We haven't been able to reproduce this without more specifics.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.

sjvudp, 2026-10-17:
Considering the fact that the report is 7 years ago (and I don't have that phone or app version any more), I think it's OK to close the bug now.