subsurface/subsurface #2180: App: Switch to numeric keyboard for numeric input

sjvudp, 2019-07-13:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Android has a feature to change the layout of the virtual on-screen layout depending on the type of input requested. Subsurface does not use that feature.
For an input field that accepts purely numeric data (for example), switching to the numeric keyboard would make input much easier.
Other types of input that have useful keyboard layouts are: Web addresses, E-Mail addresses, passwords, phone numbers, ...

### Operating system:

Android 8.0 DEU

Huawei P10 Lite

### Subsurface version:

2.1.6

True for older versions, too.

Official.

### Steps to reproduce:

1) Start adding a new dive (manually)

2) Tap on "depth" field.

### Current behavior:

Alphanumeric (standard) keyboard is being displayed

### Expected behavior:

Numeric keyboard should be displayed.

### Additional information:

### Mentions:

dirkhh, 2019-07-14:
To the best of my knowledge the toolkit that we use to create the Android app doesn't support this.

sjvudp, 2019-07-15:
>>> Dirk Hohndel <notifications@github.com> schrieb am 14.07.2019 um 02:11 in
Nachricht <Subsurface-divelog/subsurface/issues/2180/511162642@github.com>:
> To the best of my knowledge the toolkit that we use to create the Android app 
> doesn't support this.

I'm unsure, but isn't as simple as setting the HTML <input>'s "type" attribute?
type=email should set the keyboard for email, type=number, for numbers, etc.

atdotde, 2019-07-15:
A bit of googling suggests, 

https://doc.qt.io/qt-5/qt.html#InputMethodHint-enum

is relevant. We could try to set the input method hint for the input widget. Too bad I am too busy right now to check if that changes the keyboard on Android.

mikeller, 2026-10-18:
Hi @sjvudp,

Good news, this is fixed. PR #4895 "Mobile: Improve Number Entry" (merged 11 Aug 2026) added numeric keyboard support for number-only fields on mobile. I've confirmed this directly in current source: the depth field in mobile-widgets/qml/DiveDetailsEdit.qml now sets inputMethodHints: punctuationInputMethodHints(Qt.platform.os, Qt.ImhFormattedNumbersOnly), which brings up a numeric keypad on Android (and respects platform-specific punctuation conventions, e.g. decimal point vs comma).

Closing this as fixed. Please reopen if a numeric-only field still brings up the full alphanumeric keyboard on a current build.

closed 2026-10-18 by mikeller