subsurface/subsurface #2181: App lockup when switching apps

sjvudp, 2019-07-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
App locks up (fails to respond) after switching back from another app.

### Operating system:

Android 8.0 DEU

Huawei P10 Lite

### Subsurface version:

2.1.6

True for older versions.

Official.

### Steps to reproduce:

1) Start a new dive manually.

2) While in input field switch to another app (like reading an E-Mail message).

3) Switch back to subsurface.

### Current behavior:

Nothing happens for several seconds. Then Android complains the app is not responding.

### Expected behavior:

App should continue to accept input, just as other apps do.

### Additional information:

### Mentions:

dirkhh, 2019-07-14:
Interesting. I can reproduce the behavior - but I can't tell where the app is stuck.
This may be a Qt/QML problem, but the true cause is unclear to me. Something to investigate.

bstoeger, 2020-05-03:
@dirkhh: Can you still reproduce this?

dirkhh, 2020-05-03:
No, I cannot. I'm guessing our pretty complete rewrite of this part of the UI (plus a couple of Qt update since 2.1.6) might have fixed this as a side effect.
Closing.

closed 2020-05-03 by dirkhh