subsurface/subsurface #2187: NERD 2 not recognized by Android app

nolanwevans, 2019-07-20:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Android app does not recognize Shearwater NERD 2 over BLE as a dive computer. It recognizes my Perdix.

### Operating system:

Android 9

Pixel 2XL

### Subsurface version:

2.2.0

unknown

official

### Steps to reproduce:

From log file:
Found new device: "NERD 2" "LE:F0:D6:6D:37:9C:E6"
Not recognized as dive computer
Found new device: "Perdix" "LE:C9:1F:50:20:1B:C2"
"this could be a Shearwater Perdix"

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-07-20:
Thanks for sending the details  - I had never seen a NERD 2 over BLE so didn't know which BLE string to add for it. I'll fix this in the next version of the Android app.

dirkhh, 2019-07-20:
Would you please try this test build? It's signed with the release keys so you should be able to download it and install it on top of the existing app (may have to allow "Install from other sources) without losing any settings, etc.
It should recognize your NERD 2.
https://subsurface-divelog.org/downloads/test/Subsurface-mobile-4.9.0.30-arm64.apk

nolanwevans, 2019-07-20:
Works perfectly! Now if you can only speed up the dive download speed for
Shearwater computers.....

Thanks for the rapid response.

nolanwevans, 2019-07-20:
Issue resolved

commit 52cb38e5 (2019-07-25): Bluetooth: autodetect Shearwater NERD models

This should allow it to work with Subsurface-mobile as well.

Fixes #2187
closed 2019-07-20 by nolanwevans