subsurface/subsurface #2189: Shearwater Perdix gas change not handled on iOS Subsurface-mobile

iainmsmith, 2019-07-21:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Dives downloaded from Shearwater Perdix include a gas switch. This existence of the switch does appear in the download, but the actual mix is not changed. (eg a 18/45 switch to 50% displays as 18/45 to 18/45). Although this can be manually corrected in the record of the dive, the profile graph continues to show that both gases are 18/45. If the record is uploaded to the cloud and downloading it to another device, the corrected mix does not appear.

However, correcting the mix on the desktop version (4.8.6), uploading it to the cloud and downloading it to the mobile device does preserve the corrected mix.

(Of note, the Shearwater app does download the correct gases)

### Operating system:
iOS 12.3.1
iPhone 7
(also iPad Air)

### Subsurface version:
Subsurface-mobile v2.1.6 (4.8.4.66)
Unable to download Perdix to Desktop version, but see above comments re: handling of records of gas mixes.
Official release

### Steps to reproduce:
1) Download Perdix record of a dive involving a gas switch to Subsurface-mobile
2) Edit record. Correct second gas mix. Save record. Note that mix shown on profile graph has not changed.
3) Upload to cloud
4) Download from cloud. Note that corrected gas mix has not been preserved.

### Current behavior:
As described above.

### Expected behavior:
Correct gas mix should be downloaded from computer.
Manual changes to 2nd gas mix in mobile app should be preserved when uploading to cloud.

### Additional information:

### Mentions:

dirkhh, 2019-07-21:
This is odd. I haven't done multi tank dives in a while, so I haven't been able to check this recently, but this should work.
Just to make sure I understand what is going on...
- you use Subsurface-mobile 2.1.66 on iOS and download from your Sheawater Perdix
- you switch from a tank with 18/45 to an EAN50 tank, but Subsurface thinks you switched to another 18/45 tank

Which tanks are defined on the Perdix (regardless of whether they are enabled or not)? Is there more than one 18/45 tank defined?
Can you send me the Subsurface XML of one such downloaded dive (before you did any edits)?

iainmsmith, 2019-07-21:
Hi Dirk,

 

Yes – your summary is correct.

 

However, having done a fresh download into a new cloud account (to get unedited .xml) the download is now handling the gas change correctly. I don’t understand that, as the problem occurred after every one of 9 dives done last week.

 

I’ve not got another mix dive planned until late August – I’ll download the dive into my main account then and, if the problem recurs, will forward the logbook without edits.

 

Iain