subsurface/subsurface #2190: Using multiple transmitters

nolanwevans, 2019-07-21:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I use different transmitters for my singles and doubles setups with my Perdix AI and NERD 2. The doubles transmitter is registered as the second tank in both dive computers, so it (and the associated dive data) shows up as the second cylinder in the log while the first cylinder gets the O2% information. 

Would it be possible to incorporate logic to use only the second cylinder and exclude the first if the first cylinder does not have any data?

Related to this, I would like to be able to delete the first cylinder when it is unused. I cannot currently do this as it is locked.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-07-21:
That sounds wrong - but if I look at this and at #2189 I think there may a bug in the overall tank / mix handling when downloading from the Perdix (or NERD).
Can you send me an un-edited dive that you downloaded which shows the current behavior? My guess is that we don't see the information that tank 2 is the correct tank somehow...

nolanwevans, 2019-07-21:
Here you go
[dive export.zip](https://github.com/Subsurface-divelog/subsurface/files/3414715/dive.export.zip)

ryanm101, 2020-08-19:
Not sure if my issue is related:
https://github.com/subsurface/subsurface/issues/2918