subsurface/subsurface #2193: Program crashes when opening cloud storage after recent update

Gsarmedic, 2019-07-27:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Just updated to newest version. Attempted to open cloud storage. Progress bar fills then entire program closes. Attempted multiple times after reopening program from desktop shortcut. Verified that the newest updated version is the one that is active. Every attempt to open cloud storage results in program shutting itself down.

### Operating system:
Windows 10 x64
ASUS Laptop

### Subsurface version:
v4.9.0
Did not happen with previous version.
Official release.

### Steps to reproduce:

Open program. Click: File-->Open cloud storage.

### Current behavior:

Process ends and interface closes.
### Expected behavior:

Program should download log files from cloud storage and display complete list of data. AND REMAIN OPEN.
### Additional information:

### Mentions:

dirkhh, 2019-07-28:
Please try the latest continuous build.
I think this should fix this issue.

https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous

Gsarmedic, 2019-07-28:
Do I download all of the files into a folder or what? Then which one
executes the install?

mturkia, 2019-07-28:
only subsurface.exe

(the other files are for other operating systems)

dirkhh, 2019-07-28:
Well, you should be able to just use subsurface.exe and move it into an existing installation of Subsurface 4.9
Or you could use subsurface-4.9.0-xx-gnnnnnnn.exe which is a complete installer and will replace your existing installation.
Either works.

The reason why I'm not giving a direct link or a specific file name is that this gets re-created every time we update the sources, so by the time you read the email the name could have changed.

/D

Gsarmedic, 2019-07-29:
I received 4 error messages stating that it is missing .dll files.

However, I downloaded:
1. subsurface-continuous.zip
2. subsurface-4.9.0-34gb0f1b258064b.app.zip
3. subsurface-4.9.0-34gb0f1b258064b.app

I installed using the .app file and the program now functions properly.

dirkhh, 2019-07-29:
since you state that this works properly now, I'm closing the issue

closed 2019-07-29 by dirkhh