subsurface/subsurface #2199: Surface Mobile - App Crashes downloading dives

SousaMartins, 2019-07-28:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The surface mobile version 2.2.0(4.9.0.30) crashes after finish download the dives from a Shearwater perdix computer. It worked fine with the previous version

### Operating system:
Android 9

Samsung Galaxy J7

### Subsurface version:

 surface mobile version 2.2.0(4.9.0.30)

It does not happened with the previous version

I am using official release

### Steps to reproduce:

1) The Bluetooth connection is establish
2) It starts download the dives data
3) The application crashes

### Current behavior:

crashes the application

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2019-07-29:
are you able to connect to the device with `adb` (if you don't know what that means, that's ok, I'm simply asking on the off chance that you are somewhat familiar with debugging issues on an Android device.
We have a small users who report similar problems, but none of the developers experience this issue which makes it a lot harder to track down. Are you on our beta channel (i.e., do you get the more frequent updates)? I'll make a newer version available there which fixes a different potential crash, just on the off chance that that's the one you see (and you just see it with a different path leading you there)

SousaMartins, 2019-07-30:
Hi thank you for your email

 

Unfortunately I am not familiar with  debugging issues on an Android device.

I am sending you two log files I have downloaded from the Android device. I hope this will help to identify and solve the problem

 

Thanks once again

SousaMartins, 2019-07-30:
No I am not on the beta channel

dirkhh, 2019-07-30:
Since this fixed the problem for two other users, I just pushed the new build to production - you should get an update to Subsurface, soon. I would appreciate if you could report back here if that does (or, worst case, does not) fix your problem.
Thanks

SousaMartins, 2019-07-31:
Sorry the problem is not fixed.

 

I am sending you the new log file.

 

Thanks

SousaMartins, 2019-07-31:
[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/3451333/subsurface.log)

bstoeger, 2019-08-04:
Looks like a duplicate of #2167. :(

closed 2019-08-19 by SousaMartins